malnvenshorn/OctoPrint-FilamentManager #106: [Request] Archive spools

BrencelJ, 2018-06-05:
Hey, would be possible to add a button that archive spools that you used up or maybe just make different tags that you know which spools are in use and which ones are in storage.

Thanks in advance! :)

BrencelJ, 2018-06-05:
That would be great yeah. Would be possible to hide the empty ones from selection?

xefil, 2018-10-12:
Hello!
Was this implemented? A way to archive the used spools would be great as well! I would like to maintain a history of the used spools, but hide them from the view.
Thanks, Simon

onkelbeh, 2018-11-06:
and if possible: show a unique ID for every spool, so we can stick a number on them, this would clearly identify a spool. Imho you already have one for the database :-)

s1mpleman, 2018-11-17:
Has any work been done on this? Would be a great feature. Just deleting used up spools at the minute.

malnvenshorn, 2019-02-19:
I'm in the process of refactoring every part of the plugin at the moment, because I'm not happy with some decisions I've made. Hopefully there will be a new release within the next weeks. Then I will take a look at the feature requests.

xefil, 2019-02-20:
Thanks a lot :-)

NotExpectYet, 2019-02-20:
Legend! My farm thanks you ;)

BrencelJ, 2019-02-20:
That sounds great!! Thank you!

malnvenshorn, 2019-02-28:
It would be great if you could provide a precise description of what you want and how the solution should look like, ideally using the new [request template](../blob/master/.github/ISSUE_TEMPLATE/feature_request.md).

BrencelJ, 2019-03-01:
name | about | title | labels | assignees
-- | -- | -- | -- | --
Feature request | Suggest an idea for this project | Archive spools |   |  

**Is your feature request related to a problem? Please describe.**
No, it is not related to a problem per se but it would be an improvement of usability of this plugin.

**Describe the solution you'd like**
I would like to be able to archive spools that are used up so I still have a way of checking how many spools of plastics I used. I would love to be able to remove it from selections of the spools (It would be great if we could have a way to hide spools that we don't want to be able to see at the selection area).

**Describe alternatives you've considered**
Another way is to colour the spoils that we selected as empty in a different colour or that the name is crossed out from the spool selection in the main menu.

**Additional context**

malnvenshorn, 2019-03-01:
So basically it's just for some kind of statistics? 

I've never had the intention to add something like this, because that's not the purpose of this plugin. However with the next release it will be possible to filter out spools with insufficient filament for the print.

xefil, 2019-03-02:
Sounde great. I would like to avoid in cancelling ended spools. It's most a statistic use, true, btw it's a great feature to have the ability to check the number of used spools or the price I've purchased them in the past.
Thanks, Simon

xefil, 2019-04-06:
If you'll need a beta tester, let me know ;-)