malnvenshorn/OctoPrint-FilamentManager #107: Fail to re-connect to postgres server after Raspberry Pi Reboot

TheEagle13, 2018-06-08:
#### What were you doing?
Raspberry Pi lost power while I was moving equipment around. (only Pi was powered, printer was off)
Upon power back up, it can no longer connect to postgres server. The test button remains red on tests. I have also verified with a second Raspberry Pi that has no additional plugins, setup as instructed from the install instructions you provide. (Note the Raspberry Pi sits on a different subnet than the server and my computer, but I know that all routing and networking is correct)

#### What did you expect to happen?
Complete connection and show spool inventory like normal
#### What happened instead?
No inventory displayed, connection attempts fail
#### Link to octoprint.log

Attached
[octoprint.log](https://github.com/malnvenshorn/OctoPrint-FilamentManager/files/2082653/octoprint.log)

#### Link to gcode file

N/A

#### Link to contents of Javascript console in the browser

N/A?

#### Further information

I have discovered that when this happens I am able to recover by disabling the FilamentManager plugin from the octoprint plugin manager, then re-enabling, however this requires 2 octoprint service reboots, and takes extra time.

I have read the FAQ and provided all required information.

malnvenshorn, 2018-06-08:
First it doesn't seem to be a bug.

> ERROR - Failed to initialize database: (psycopg2.OperationalError) could not connect to server: Network is unreachable

The plugin simply fails to connect because the PostgreSql server is not reachable at the time the plugin is loaded and there is nothing I can do about it.

Because the plugin only tries to connect at startup you have to restart OctoPrint to give it another try. That might be something I could change, either by automatically start a retry or by presenting a dialog for the user to manually start a connection attempt .

TheEagle13, 2018-06-08:
Problem is, I did try to restart octoprint as I assumed that was part of the problem, and It still wouldn't connect. The only resolution was to disable and re-enable the plugin. I believe this is a bug, as the issue only appears to be with the plugin, since the postgres server is still available on the network. So Is the plugin attempting to connect before the Raspberry Pi starts it's networking services?

malnvenshorn, 2018-06-08:
The log file doesn't show any connection error after you restarted OctoPrint the first time. What makes you assume that there was no database connection? Have you reloaded the web frontend after the restart?

> So Is the plugin attempting to connect before the Raspberry Pi starts it's networking services?

This would be an explanation for the issue. If you are using systemd to start OctoPrint you can specify something like 
```
[Unit]
Wants=network-online.target
After=network-online.target
```
in the service file to make sure OctoPrint will be started after the network is ready.

TheEagle13, 2018-06-09:
Forgive me if I'm mistaken, and correct me if I'm wrong, but wouldn't it be a bug if it is something that is an undesired effect, that requires extra effort to resolve? 

Don't get me wrong, I love your work so much that I will gladly jump through hoops every startup just to be able to use your plugin, but in my thought process... I shouldn't have to if there wasn't a bug?

I honestly have no clue how Octoprint is started because I never bothered to look into it. I'm running the OctoPi image, so I didn't build myself.

malnvenshorn, 2018-06-09:
Well, a bug is a software or system error that causes incorrect or unexpected results or produces unexpected behavior. This is how I would define it.

But in your case the plugin behaves as it should be. The connection process fails on startup because of something unrelated to this plugin (network or server not available). In such case you have to restart OctoPrint so that the plugin tries to connect again.

As I wrote before I could improve the connection process to make it unnecessary to restart the OctoPrint instance after the connection fails. But this won't solve your problem, because it's not cause by the filament manager. 

> I honestly have no clue how Octoprint is started because I never bothered to look into it. I'm running the OctoPi image, so I didn't build myself.

Me neither. I don't run the OctoPi image.

malnvenshorn, 2018-06-10:
If there is a bug somewhere then you have to take a look at the `psycopg2` module. This library is used for the PostgreSql  connection. All I do is to pass the server URI and credentials. If the library fails to make a connection (with the error from your log file) then it is very likely a problem with your server or network connection. That's all I can say.

malnvenshorn, 2018-06-10:
> Because Octoprint and any other plugin I have (that connects to the internet) has absolutely no problem on startup.

Wrong!

`2018-06-07 23:29:46,591 - octoprint.startup - INFO - We don't appear to be online, not fetching plugin blacklist`

> If that's where the bug is, then why would restarting octoprint resolve the issue?

Because it tries to connect again.

closed 2018-06-10 by malnvenshorn