malnvenshorn/OctoPrint-FilamentManager #110: "[Question]" About setting up remote database

mrwizerd, 2018-07-01:
I have gone through the tutorial for setting up the remote database and I am pretty sure I have everything running properly.  Currently the server is a Raspberry Pi 3+ with a fresh install of Rasbian latest release.  I then installed Postgresql which took some figuring out because some of the commands didn't work, for instance I had to run the initdb by using the full path to it.   This worked and I was able to initialize the database after manually creating the directory.  Then I started and enabled the service, created the octoprint user and filamentmanager database as per the instructions.  So the only thing that was modified was the 
`sudo pacman -Sy postgresql`
to
`sudo apt-get install postgresql`
and added the full file path for initdb to
`$ sudo -u postgres -i initdb --locale $LANG -E UTF8 -D /var/lib/postgres/data`
Anyways everything after that was as per the instructions.  

I restarted the PostgreSQL service at the end and then went to my octoprint install on the raspberry pi 3 ssh to it and opeend the vertialenv and ran
`pip install psycopg2`
per the instructions, then restarted octoprint, and went to the filament manager and  pointed the database to postgresql://192.168.42.101 (raspberry pi 3+) with the database, user name as stated and the password I gave it.  I then restarted octoprint per the note on that page and tested the database.  Nothing happened no green button, no error.  So i tried importing and it errors telling me to check the log.  I checked the log and the log said

`2018-07-01 03:38:44,721 - octoprint.plugins.filamentmanager - ERROR - Data import failed: 'NoneType' object has no attribute 'import_data'
2018-07-01 03:38:44,732 - tornado.access - ERROR - 500 POST /plugin/filamentmanager/import (127.0.0.1) 44.85ms
2018-07-01 03:38:57,217 - tornado.access - WARNING - 400 POST /plugin/filamentmanager/database/test (127.0.0.1) 32.64ms
2018-07-01 03:38:58,318 - tornado.access - WARNING - 400 POST /plugin/filamentmanager/database/test (127.0.0.1) 27.64ms
2018-07-01 03:39:04,082 - tornado.access - WARNING - 400 POST /plugin/filamentmanager/database/test (127.0.0.1) 26.20ms
2018-07-01 03:39:12,100 - octoprint.plugins.filamentmanager - ERROR - Data import failed: 'NoneType' object has no attribute 'import_data'
2018-07-01 03:39:12,113 - tornado.access - ERROR - 500 POST /plugin/filamentmanager/import (127.0.0.1) 44.32ms
2018-07-01 03:39:31,465 - octoprint.plugins.filamentmanager - ERROR - Failed to create profile: 'NoneType' object has no attribute 'create_profile'
2018-07-01 03:39:31,477 - tornado.access - ERROR - 500 POST /plugin/filamentmanager/profiles (127.0.0.1) 32.89ms
2018-07-01 03:47:19,648 - tornado.access - WARNING - 400 POST /plugin/filamentmanager/database/test (127.0.0.1) 30.53ms
2018-07-01 03:47:20,611 - tornado.access - WARNING - 400 POST /plugin/filamentmanager/database/test (127.0.0.1) 26.36ms
2018-07-01 03:47:21,518 - tornado.access - WARNING - 400 POST /plugin/filamentmanager/database/test (127.0.0.1) 26.09ms
2018-07-01 03:47:22,277 - tornado.access - WARNING - 400 POST /plugin/filamentmanager/database/test (127.0.0.1) 27.47ms
2018-07-01 03:47:28,159 - tornado.access - WARNING - 400 POST /plugin/filamentmanager/database/test (127.0.0.1) 26.91ms
2018-07-01 03:47:34,258 - octoprint.plugins.filamentmanager - ERROR - Data import failed: 'NoneType' object has no attribute 'import_data'
2018-07-01 03:47:34,271 - tornado.access - ERROR - 500 POST /plugin/filamentmanager/import (127.0.0.1) 46.21ms
`

Which brings us to the question, do you know if this is complaining that it can not access the database server, or is it complaining about the user?  I am not very savy with linux though I am pretty literate and am good at learning new things but I am unsure how to diagnose the situations .  I can provide further details on request.  Just let me know how I can help you help me.

I have reset both the 3 and the 3+ Octoprint and SQL Server this did not solve the problem,  I have a feeling the problem is either with octoprint logging into the SQL Server (I did modify the address tables) as the log is using local host address... but I may be incorrect, also though I installed the required dependency I am not positive it is working as I am unsure as to how to check  that.

malnvenshorn, 2018-07-01:
Have you modified the `postgresql.conf` and `pg_hba.conf` according to the wiki?

> Which brings us to the question, do you know if this is complaining that it can not access the database server, or is it complaining about the user?

I need the full log file to answer this question.

mrwizerd, 2018-07-02:
Yes I did modify those files.  Is it ok to post the full log here in text,
or like at paste bin or something?  I will post it later today.

malnvenshorn, 2018-07-02:
You can simply drag and drop the file here and it will be added as an attachment.

mrwizerd, 2018-07-14:
Sorry I was on vacation for independence day and have been catching up I
will upload the log in the next 24 hours

mrwizerd, 2018-07-15:
Sorry for the delay I had to recreate the attempt, I initially set it up
and didn't have the other pi on, so I redid the setup with the other pi
on.  so there may be two entries in the log.  The last boot will be the
proper attempt.

malnvenshorn, 2018-07-23:
No problem. I'll reopen this ticket when you upload your log file.

mrwizerd, 2018-07-28:
[octoprint.log](https://github.com/malnvenshorn/OctoPrint-FilamentManager/files/2237679/octoprint.log)
sorry I thought I uploaded this with my last reply.

malnvenshorn, 2018-07-29:
> 2018-07-15 13:18:56,474 - octoprint.plugins.filamentmanager - ERROR - Failed to initialize database: No module named psycopg2

You haven't read the entire wiki entry. You have to install the missing python module.

mrwizerd, 2018-07-29:
> I restarted the PostgreSQL service at the end and then went to my octoprint install on the raspberry pi 3 ssh to it and opeend the vertialenv and ran
pip install psycopg2
per the instructions, then restarted octoprint, and went to the filament manager and pointed the database to postgresql://192.168.42.101 (raspberry pi 3+) with the database, user name as stated and the password I gave it. I then restarted octoprint per the note on that page and tested the database. Nothing happened no green button, no error. So i tried importing and it errors telling me to check the log. I checked the log and the log said

I actually did, the wiki says to run

> $ ~/OctoPrint/venv/bin/pip install psycopg2

Which I tried and it returns 

> -bash: pip: command not found

So I looked it up and went into venv and ran it and I thought it installed, though to be sure at this point I was winging it.  With the confirmation that it had not installed now I have gone and found another website to install pip [https://www.tecmint.com/install-pip-in-linux/](url) here and have run it.  I then ran the install for psycopg2 and it ignored some stuff and said it was successful.  Tested, failed.  Went back in and tried to install psycopg2 again and this time it didn't give the ignore statement.  Tested again, failed.

Posting log file. it looks like it is still not able to initialize due to lacking psycopg2 but it was installed.

> JadeMaker:~/OctoPrint/venv/bin $ ~/OctoPrint/venv/bin/pip install psycopg2
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: psycopg2 in /home/pi/OctoPrint/venv/lib/python2.7/site-packages (2.7.5)

that shows int installed in the lib folder though not the bin folder which the command used.  I don't know if this is the incorrect folder or what.

[octoprint.log](https://github.com/malnvenshorn/OctoPrint-FilamentManager/files/2239120/octoprint.log)

mrwizerd, 2018-07-29:
I apologize for the problems, and thank you profusely for the help.  I am unsure if I should just reinstall everything from scratch at this point again or what.  The learning curve is pretty steep when all the commands dont execute the same way as the wiki which is just due to upgrades in the os I am sure.

malnvenshorn, 2018-07-30:
You have to install the psycopg2 module to the virtual environment you are using for your OctoPrint instance. If you followed the installation guide linked in the wiki the path should be `~/OctoPrint/venv/bin/pip`. Otherwise you need to adjust the path according to your installation.

malnvenshorn, 2018-08-18:
I'm closing this ticket now because there has been no further comments for some time.

closed 2018-08-18 by malnvenshorn