malnvenshorn/OctoPrint-FilamentManager #114: Unable to import

Jonnie-Cache, 2018-07-19:
#### What were you doing?

Performed an export. Unzipped. Modified CSV files. Zipped. Attempted upload.

#### What did you expect to happen?

Upload would be successful.

#### What happened instead?

Received an error message:

Data import failed
Something went wrong, please consult the logs.

#### Link to octoprint.log

[LOG](https://d.pr/f/bPEijt)

#### Link to gcode file

N/A

#### Link to contents of Javascript console in the browser

[LOG](https://d.pr/f/SZvXro)

#### Further information

Nothing I can think of.

I have read the FAQ and provided all required information.

malnvenshorn, 2018-07-23:
> Data import failed: [Errno 2] No such file or directory: '/tmp/tmpu7AhCe/profiles.csv'

Looks like you didn't pack the profiles.csv into the archive.

Jonnie-Cache, 2018-07-23:
I did. Here is the ZIP file that I used.

[import.zip](https://d.pr/f/4t32R6)

malnvenshorn, 2018-07-24:
You packed the files inside a sub folder. Take a look at the exported file how it should look like.

Jonnie-Cache, 2018-07-24:
Ah! I'll give that a try. From my recollection, when I exported and unzipped the files, they came in a subdirectory already. I believe that I just edited them in place and re-zipped them. I'll try again tonight.

Jonnie-Cache, 2018-07-26:
That worked, but FYI, when I exported the files again, the unzipped in a sub directory again.

malnvenshorn, 2018-07-26:
If there is a sub directory you should blame your extraction software. Closing.

closed 2018-07-26 by malnvenshorn

Jonnie-Cache, 2018-07-26:
My apologies. I had no idea that extraction software would make a difference. If I may make a suggestion for enhancement, it would be beneficial if the software could do one or more of the following:

1. Provide more meaningful error messages when the .ZIP file contains a sub-directory.
2. Provide on-screen information/instruction to let the user know that the uploaded .ZIP file must have the files in the root of the archive file.
3. Update documentation to inform the user of the requirements for the .ZIP file.

Thanks for your help!