malnvenshorn/OctoPrint-FilamentManager #120: Multiple Octopi servers always use the same last selected spool

JamesVexica, 2018-08-22:
Before opening a new ticket please take a look if there is already an existing one addressing the same concern.

Also read the FAQ: https://github.com/malnvenshorn/OctoPrint-FilamentManager/wiki#faq

If you have a **request** or **question** mark them with a "[Request]" respectively "[Question]" prefix in the title.

For **bug** reports use the template below and provide ALL required information. Do NOT delete ANY lines from the template but those enclosed in [ and ]. Also make sure you are using the latest version and that you have read the release notes.

(Before submitting your ticket, please delete this text up to and including the line too - it's only here for your information)

---

#### What were you doing?
Setup postgresql to use database for filament spools. 
Select a spool and start a print on server 1. 
Go to server two and select a different spool for use. 
Go back to server 1 and hit refresh and it loads the spool selected for server two. 

#### What did you expect to happen?
Server 1 should keep the spool selected, and ignore the selected spool of server two. 

#### What happened instead?
Server 1 changed it's selected spool to that selected on server 2. 
Server 1 also deducts the used filament from the spool that was selected in server 2.

#### Link to octoprint.log

[Always include the log file.]

#### Link to gcode file

[If you encounter issues with the odometer feature provide a small gcode file which produces the problem.]

#### Link to contents of Javascript console in the browser

[If you're reporting UI issues always include the output of the javascript console of your browser. Always include if unsure.]

#### Further information

I will post logs as soon as this 50 hour print I've started is done. I'll setup two small prints to make sure it's doing what I'm seeing. 
It's certainly deducting from whatever spool is selected second as it told me I'd used a full roll of wood filament with only a few small prints on the server that was using it, and one large print on the second server that was using a different spool. 

I have read the FAQ and provided all required information.

malnvenshorn, 2018-08-23:
Every OctoPrint instance should have a unique client ID to prevent your described behavior. Could you check if those IDs differ from each other? You can find them in the `config.yaml` under the path `plugins > filamentmanager > database > clientID`.

JamesVexica, 2018-08-23:
Found it! Your correct. Sorry about that false positive, I'd forgotten when I setup the pi's I installed everything on one instance and copied the SD card to the next. I'll remember to change that ID when I add another printer instance :). 

For anyone else with a similar issue, I just changed the last two digits of the Client ID so they are both unique. I've not had chance to check it, but both printers are respecting the choices made on the filament now. Cheers! Thanks for a great plugin too, works a treat.

closed 2018-08-23 by JamesVexica