malnvenshorn/OctoPrint-FilamentManager #122: [Request] Add flow offset for spools

kvuorine, 2018-09-04:
How about adding flow offsets for spools for easier management of proper flow rates?

monterman, 2018-11-12:
I also would support this, for my Marlin printer using linear advance, it would be great to have a M900 Kxx to adjust the linear advance K factor tweaked to different material spools.

malnvenshorn, 2019-03-05:
Do you want this feature to print different materials or is there another reason?

@monterman I guess the value is only depending on the material?

remy561, 2019-03-08:
I'd like this addition as well. Most of my spools print great at .92% flow in CURA. However, one of my spools requires .98% as it is probably a bit too thin. Being able to set the Octoprint flow rate to these settings instead of CURA allows me to slice a model without taking the spool into account :)

kvuorine, 2019-03-08:
I suggested this feature for exactly the same reason as remy561.

JranZu, 2019-06-10:
I'd also like to place my vote for this. I've got my Cura settings dialed in -- most of the time all I need to do is offset the temperature and I can use the same file for multiple spools, however a few of my spools needs to print at 105%. I hate having to create a new slice just for those spools.

rescuehero, 2019-09-05:
would also like this feature, because i got an extremely wide range of flowrates for different filamentmanufacturers.