malnvenshorn/OctoPrint-FilamentManager #124: Actual Remote Database - 400 Post error

jwynn6, 2018-09-14:
#### What were you doing?

Connecting Octopi based Filament Manager to an externally hosted Postgresql 10 database.  Also tried with a Postgres 9.5 database.  I have ensured that other systems (ubuntu) are able to connect remotely to the same database with the same username and password. The Octopi has psycopg2 installed as directed on the wiki.

#### What did you expect to happen?

Test button to turn green.

#### What happened instead?

Turned red and stayed red.  Throws a Warning  - 400 Post error in the log.

#### Link to octoprint.log

[octoprint.log](https://github.com/malnvenshorn/OctoPrint-FilamentManager/files/2384877/octoprint.log)

#### Further information

I have read the FAQ and provided all required information.

Stady234, 2018-09-15:
I am trying to validate the octoprint side of the plugin to the archlinux database on a seperate vm.  Connecting the pgadmin4 from my desktop to the database works fine.  When i click the test button in the plugin... it stays red.  I had to use [sudo apt-get install python-psycopg2] to get it to install as your documented way states there is config files missing.  Where do i get a log you need to see what is going on?

malnvenshorn, 2018-09-15:
`octoprint.plugins.filamentmanager - ERROR - Failed to initialize database: Engine 'postgres' not supported`

Your database URI is wrong, it should start with `postgresql://`. See the wiki on how to set it up correctly.

malnvenshorn, 2018-10-01:
I'm closing this ticket now because there has been no further response for some time.

closed 2018-10-01 by malnvenshorn