malnvenshorn/OctoPrint-FilamentManager #125: [Question] Sync delay? What is normal?

Stady234, 2018-09-15:
---

#### What were you doing?

Able to insert single spool and see replication across multiple pi's.  When i Import the master list, the pi i used shows the list.. the others after sitting all night still dont show the new list.

#### What did you expect to happen?
import to update the database
#### What happened instead?
only the pi used to import shows the update others, even after restart, do not show the update.
#### Link to octoprint.log
1-octoprint.log = main pi used to import the master list
2-octoprint.log = secondary pi only showing the manual entered spool

https://www.dropbox.com/sh/nnn1kiucyptdigr/AAAY6U9ebYGTonzTvnVHpGtJa?dl=0

#### Link to gcode file

nothing used...

#### Link to contents of Javascript console in the browser

No UI issues that I am aware of..

#### Further information

I have restarted all devices multiple times... all the pi's; the db vm; my computer; the computer the vm is on.... still no updates after import.

I have read the FAQ and provided all required information.

malnvenshorn, 2018-09-15:
It seems you have some configuration error you should check first, e.g.

`octoprint.plugins.filamentmanager - ERROR - Failed to initialize database: Engine 'postgres' not supported`

`octoprint.plugins.filamentmanager - ERROR - Failed to initialize database: (psycopg2.OperationalError) FATAL:  no pg_hba.conf entry for host "10.0.1.22", user "octoprint", database "octoprint_filamentmanger", SSL off`

malnvenshorn, 2018-10-01:
I'm closing this ticket now because there has been no further response for some time.

closed 2018-10-01 by malnvenshorn