malnvenshorn/OctoPrint-FilamentManager #127: Temperature goes to 210 instead of 200, no offsets

ikoutone, 2018-10-08:
#### What were you doing?
Using Cura 3.5.0. Bed is set to 60C and print temperature is set to 200C in Cura.

#### What did you expect to happen?
Print would start normally with the temperatures defined in Cura.

#### What happened instead?
Print started with print temperature set to 210C.

#### Link to octoprint.log

2018-10-08 13:07:42,511 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Detecting serial port"
2018-10-08 13:07:42,798 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting serial port" to "Opening serial port"
2018-10-08 13:07:42,806 - octoprint.util.comm - INFO - Changing monitoring state from "Opening serial port" to "Detecting baudrate"
2018-10-08 13:07:43,171 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.0.21) 18.87ms
2018-10-08 13:07:43,832 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2018-10-08 13:07:45,167 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.0.21) 18.72ms
2018-10-08 13:07:47,171 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.0.21) 18.93ms
2018-10-08 13:07:49,166 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.0.21) 18.30ms
2018-10-08 13:07:51,187 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.0.21) 21.65ms
2018-10-08 13:07:51,684 - octoprint.filemanager.analysis - INFO - Starting analysis of local:CCR10_4mm__glass_bed_adapter.gcode
2018-10-08 13:07:51,689 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/oprint/bin/python2 -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 /home/pi/.octoprint/uploads/CCR10_4mm__glass_bed_adapter.gcode
2018-10-08 13:07:53,171 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.0.21) 20.91ms
2018-10-08 13:07:55,173 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.0.21) 20.48ms
2018-10-08 13:07:55,597 - octoprint.plugins.portlister - INFO - do_auto_connect
2018-10-08 13:07:55,599 - octoprint.plugins.portlister - INFO - realpath match
2018-10-08 13:07:55,688 - octoprint.plugins.portlister - INFO - Not autoconnecting; printer already connected
2018-10-08 13:07:57,170 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.0.21) 20.44ms
2018-10-08 13:07:59,174 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.0.21) 21.77ms
2018-10-08 13:08:00,326 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2018-10-08 13:08:01,180 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.0.21) 20.31ms
2018-10-08 13:08:01,331 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2018-10-08 13:08:02,338 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2018-10-08 13:08:03,172 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.0.21) 20.06ms
2018-10-08 13:08:03,351 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2018-10-08 13:08:04,364 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2018-10-08 13:08:04,378 - octoprint.util.comm - INFO - Changing monitoring state from "Detecting baudrate" to "Operational"
2018-10-08 13:08:04,391 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2018-10-08 13:08:04,433 - octoprint.util.comm - INFO - Printer reports firmware name "Marlin 1.1.9 (Github)"
2018-10-08 13:08:04,446 - octoprint.util.comm - INFO - Firmware states that it supports temperature autoreporting
2018-10-08 13:08:12,327 - octoprint.filemanager.analysis - INFO - Analysis of entry local:CCR10_4mm__glass_bed_adapter.gcode finished, needed 20.64s
2018-10-08 13:08:20,768 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Printing"
2018-10-08 13:08:20,787 - octoprint.plugins.pushover - INFO - Event triggered: PrintStarted
2018-10-08 13:08:20,800 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2018-10-08 13:08:22,900 - octoprint.util.comm - INFO - Printer seems to support the busy protocol, telling it to set the busy interval to our "communicationBusy" timeout - 1s = 2s
2018-10-08 13:08:27,786 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Cancelling"

#### Link to gcode file
https://1drv.ms/u/s!AukfbqFAaScThY53rSY_cdlOnE2AWw

#### Further information
I haven't set any temperature offsets to any filaments I have added to the manager. Temperature is always 210C in octoprint. The prints have been made with one filament though.

Also on another issue, cost is not shown on the main octoprint page while I have added the prices of filaments to the spools.

Screenshot from the filament spool profile:
https://i.imgur.com/HKc1ew8.png

I have read the FAQ and provided all required information.

malnvenshorn, 2018-10-08:
This has nothing to do with this plugin. Besides that there is no issue at all, in the gcode file: `M109 S210 ; wait for extruder temp`. So the temperature is set correctly.

closed 2018-10-08 by malnvenshorn

ikoutone, 2018-10-08:
It started happening after I installed this plugin, so that's odd timing wise then.

malnvenshorn, 2018-10-08:
Yes maybe, but this plugin has nothing to do with cura and as I sad before, the temp is set correctly according to the gcode file.

ikoutone, 2018-10-08:
Okay, yeah. Sorry. Looks like it is indeed a problem with 3.5 cura https://github.com/Ultimaker/Cura/issues/4469