malnvenshorn/OctoPrint-FilamentManager #128: [Request] FAQ answer about the sidebar ordering not working

tcurdt, 2018-10-09:
#### What were you doing?

I am trying have the filament manager below the state on the left.
(see FAQ)

#### What did you expect to happen?

I modified the `config.yaml` according to the FAQ

```
...
appearance:
  components:
    order:
      sidebar:
      - connection
      - state
      - plugin_filementmanager
      tab:
      - temperature
...
```
and then I restarted the service.
(Shouldn't there als be the file section in the sidebar config?)

#### What happened instead?

Unfortunately the order has not changed.

#### Link to octoprint.log

[octoprint.log](https://github.com/malnvenshorn/OctoPrint-FilamentManager/files/2458642/octoprint.log)

#### Link to gcode file

#### Link to contents of Javascript console in the browser

#### Further information

I have read the FAQ and provided all required information.

tcurdt, 2018-10-09:
@malnvenshorn FWIW: "works for me" has never been a very helpful attitude in the context of resolving issues. As an open source developer I understand where you are coming from but since you added the FAQ entry you kind of opened that box (or rather can of worms?). It's usually a good idea to at least point people into the right direction.

malnvenshorn, 2018-10-09:
Yes a agree the "works for me" usually sucks. But in the FAQ entry you can also find a link to the OctoPrint documentation where I got the information. The snippet is correct and works. If you have issues it's a problem with OctoPrint and not related to this plugin. Therefor you should ask in the OctoPrint forum or open a ticket on OctoPrints' bug tracker.