malnvenshorn/OctoPrint-FilamentManager #129: [Question] Multiple OctoPrint instances, single spool

stelgenhof, 2018-10-14:
Tracking the usage with your plugin works fantastic. 

Just was wondering about the situation when you use the same spool but with different printers (i.e. OctoPrint instances). 

I know the plugin can be configured to have a remote database. Would it cause any issue when multiple instance of OctoPrint with this plugin access that database? Concurrency is not the issue, just not sure about the filament usage being updated correctly by multiple instances.

Cheers! Sacha

moclub4, 2018-10-30:
Very interesting question, same situation here with me. Two printers, two octoprint instances, sharing spools. How to handle? I would be happy if there are any solutions.

monterman, 2018-11-12:
I support this feature also.  I noticed that the actual plugin is intalled in a shared directory for my two instances of octoprint, but the inventory is different.  Where is the config file for the inventory?

malnvenshorn, 2018-11-13:
There is no issue if you use an external database system. When you share the embedded SQLite database with multiple instances problems might arise if the database is already locked by another process.

stelgenhof, 2018-11-13:
@malnvenshorn Thanks for the explanation. Makes sense. :)

closed 2018-11-13 by stelgenhof