malnvenshorn/OctoPrint-FilamentManager #130: Support duplicate mode, multiple spools being used at the same time, same volume

jim-edwards, 2018-10-15:
My printer has support for duplicate mode and mirroring mode.  Basically 2 nozzles run at the same time, but the build platform is cut by the number of nozzles used (1/2 for 2).

From the G-Code, the use of duplicate mode can be detected by the M605 command, specifically M605 S2
http://marlinfw.org/docs/gcode/M605.html.

When in duplicate mode, the spools for toolheads T0 and T1 should both be used equal amounts.

malnvenshorn, 2019-02-28:
How does this duplicate mode work? You simply enable it with `M605 S2`  and the printer firmware will use every gcode command for both extruders? Does OctoPrint support this (shows the length for both extruders)? Since I don't have a dual extruder I have no experience with it.

jim-edwards, 2019-04-08:
Yes, Enter M605 S2 and the printer firmware automatically does every gcode with both extrudes at the same time.  They are typically on the same Y axis and obviously Z axis, so the X motors move in unison.  It cuts the build volume in half in that direction as the heads are offset by exactly 1/2 from each other.  Some youtube videos that show the setup: https://www.youtube.com/watch?v=8N4H9zN2VfE  and https://www.youtube.com/watch?v=0J0Y5gtCsQ0