malnvenshorn/OctoPrint-FilamentManager #134: Remote Database Initialization Error

bleem313, 2018-11-15:
Before opening a new ticket please take a look if there is already an existing one addressing the same concern.

Also read the FAQ: https://github.com/malnvenshorn/OctoPrint-FilamentManager/wiki#faq

If you have a **request** or **question** mark them with a "[Request]" respectively "[Question]" prefix in the title.

For **bug** reports use the template below and provide ALL required information. Do NOT delete ANY lines from the template but those enclosed in [ and ]. Also make sure you are using the latest version and that you have read the release notes.

(Before submitting your ticket, please delete this text up to and including the line too - it's only here for your information)

---

#### What were you doing?

After setting up a remote database, every time the server starts the plugin throws an error. Everything appears to still work correctly. I am able to add and remove rolls. However, I don't know if some of the functionality is broken because of it.

#### What did you expect to happen?

The server to start without the plugin manager throwing an error.

#### What happened instead?

The plugin throws an sql error stating it failed to initialize the database.

#### Link to octoprint.log

[octoprint.log](https://drive.google.com/open?id=172yUXlcHKN5sJ0BaHpYViEAD9o3E6uRw)

#### Link to gcode file

N/A

#### Link to contents of Javascript console in the browser

N/A

#### Further information

I have read the FAQ and provided all required information.

malnvenshorn, 2018-11-15:
You need at least PostgreSQL 9.5 as stated in the README.

Duplicate of #96

bleem313, 2018-11-15:
That fixed it. Thank you for the quick reply. I forgot that the server I'm on is an older server, and didn't think to check version numbers.

closed 2018-11-15 by bleem313