malnvenshorn/OctoPrint-FilamentManager #135: [question] mysql server on NAS

dogson3006, 2018-11-23:
hi there,
i´ve got a second printer and a second raspberry pi.
so my idea is to share the database for both raspi´s.
but i won´t install a postgresql on one of these raspi, because i have a NAS running a mysql 5 server.
so, it is possible to share the database above my NAS? and if the answer is yes, how will i do this?

thanks 4 help
volker

malnvenshorn, 2018-11-23:
Only PostgreSql is supported, because MySql lacks the support to notify clients about database changes.

dogson3006, 2018-11-23:
i´m sad to hear this. 
but can you please tell me how to install pstgresql on my raspberry pi 3+ running the image from octopi on it?
only typing pip install psycopg2 over putty doesn´t work
thxx

malnvenshorn, 2018-11-23:
I don't use OctoPi. You can find a guide for Arch Linux in the wiki.

dogson3006, 2018-11-29:
thank you for answering

closed 2018-11-29 by dogson3006