malnvenshorn/OctoPrint-FilamentManager #136: Temperature offset not set at start

k4b3z4, 2018-11-30:
#### What were you doing?

1) octoprint start
2) the predefined spool (last used) has a temperature offset of -10
3) the temperature offset of the interface (temperature tab) remains at 0
4) If I select another spool and return to the original, the offset of the interface changes correctly to -10

#### What did you expect to happen?

1) octoprint start
2) the predefined spool (last used) has a temperature offset of -10
3) the temperature offset of the interface must be at -10

#### What happened instead?

when start octoprint the spool pre-selected does not change the temperature offset corresponding to the active spool.

#### Link to octoprint.log

[Always include the log file.]

#### Link to gcode file

[If you encounter issues with the odometer feature provide a small gcode file which produces the problem.]

#### Link to contents of Javascript console in the browser

[If you're reporting UI issues always include the output of the javascript console of your browser. Always include if unsure.]

#### Further information

[If you have more information which could help me to understand/fix the issue feel free to add them here.]

I have read the FAQ and provided all required information.

k4b3z4, 2018-12-14:
apparently this happens only with the release candidate of octoprint, I suppose due to some javascript problem in the octoprint interface.

closed 2018-12-14 by k4b3z4

malnvenshorn, 2018-12-14:
So it is/was(?) an issue within OctoPrint? Hadn't had time to investigate yet.

k4b3z4, 2018-12-26:
yes, apparently release 1.3.10 solved the problem. thanks!