malnvenshorn/OctoPrint-FilamentManager #138: Can't select spool / start print, no cost estimation visible

Freestila, 2018-12-18:
#### What were you doing?

Installed FilamentManager on new Octoprint, created Profile and two spools (one already with less filament).
Printed two small models - remaining filament was not reduced.
Activated config "ask for spool before print", started a new print. Selected spool.

#### What did you expect to happen?
I should be able to start the printjob. The filament should be reduced even without the "select a spool" option.

#### What happened instead?
I could not start the print, the start button was inactive in spool select window.
Without the spool select option enabled, i can start a print, but filament is not reduced.

Also i don't see the cost in the sidebar.
#### Link to octoprint.log

attached
[octoprint.log](https://github.com/malnvenshorn/OctoPrint-FilamentManager/files/2690462/octoprint.log)

#### Link to contents of Javascript console in the browser

[If you're reporting UI issues always include the output of the javascript console of your browser. Always include if unsure.]

#### Further information

i have the fllowing additional plugins installed:

[image]

I have read the FAQ and provided all required information.

Freestila, 2018-12-19:
So, i don't know why, but after the third or so restart at least partially it is working. (i did restarted everything after installing the plugin the first time). I can select the spool in the bottom and it is subtracting the used filament. Still no print cost calculation, but i could live without that. I think you can close tis ticket for now :) Thx for the good work.

malnvenshorn, 2018-12-20:
You have to select a spool, otherwise it won't work.

```
2018-12-18 10:21:21,266 - octoprint.plugins.filamentmanager - INFO - Filament used: 75.74586 mm (tool0)
2018-12-18 10:21:21,417 - octoprint.plugins.filamentmanager - WARNING - No selected spool for tool0
```

If you use the confirmation dialog you also have to wait till OctoPrint finishes the analyzing process before you can confirm your selected spool. Depending on the model it can take quite a while.

The cost estimation is not a feature of this plugin. You have to install the [CostEstimation-Plugin](https://github.com/malnvenshorn/OctoPrint-CostEstimation)

closed 2018-12-20 by malnvenshorn

Freestila, 2019-01-04:
Ah ok, i just saw this in one of the screenshots for this plugin :) so i thought it would be part of the filament manager (since the filament manager knows the spool price and weight it would be possible to show the cost for the selected spool).

malnvenshorn, 2019-01-06:
Yes, I thought about joining both plugins, but decided to keep them separate for users who want to use the CostEstimation plugin without the FilamentManager.