malnvenshorn/OctoPrint-FilamentManager #141: Plugin Not Working Properly with Multiple Toolheads

kietcallies, 2019-02-03:
Before opening a new ticket please take a look if there is already an existing one addressing the same concern.

Also read the FAQ: https://github.com/malnvenshorn/OctoPrint-FilamentManager/wiki#faq

If you have a **request** or **question** mark them with a "[Request]" respectively "[Question]" prefix in the title.

For **bug** reports use the template below and provide ALL required information. Do NOT delete ANY lines from the template but those enclosed in [ and ]. Also make sure you are using the latest version and that you have read the release notes.

(Before submitting your ticket, please delete this text up to and including the line too - it's only here for your information)

---

#### What were you doing?
I own a 4-head Stacker 500 which I control via OctoPi. I am able to set up the profiles and spools. I load the spool into the toolhead without any issues. However, when the print is completed, the toolheads are blank and the spools show no filament usage.

[Be as specific as possible. Providing a step by step guide will help me to reproduce the issue. This is needed in order to fix the problem. I cannot help you if I don't know what exactly you did to trigger the issue.]

#### What did you expect to happen?
The toolheads to display the spools assigned to them with the appropriate amount of filament subtracted from the spool for the toolhead that was used.

#### What happened instead?

#### Link to octoprint.log

[Always include the log file.]

#### Link to gcode file

[If you encounter issues with the odometer feature provide a small gcode file which produces the problem.]

#### Link to contents of Javascript console in the browser

[If you're reporting UI issues always include the output of the javascript console of your browser. Always include if unsure.]

#### Further information

[If you have more information which could help me to understand/fix the issue feel free to add them here.]

I have read the FAQ and provided all required information.

I own a 4-head Stacker 500 printer controlled by OctoPi. I am able to load profiles and spools with no issue. However, after finishing a print, the spool assignments to the toolheads go blank and not filament is subtracted from the spool. Looking octoprint.log at the last print, I am seeing this error:

2019-02-03 10:46:56,331 - octoprint.plugins.filamentmanager - INFO - Filament used: 67439.62 mm (tool0)
2019-02-03 10:46:56,355 - octoprint.plugins.filamentmanager - ERROR - Failed to update filament on tool0: (sqlite3.OperationalError) no such column: selections.client_id [SQL: u'SELECT selections.tool, selections.client_id, selections.spool_id, spools.id, spools.profile_id, spools.name, spools.cost, spools.weight, spools.used, spools.temp_offset, profiles.id, profiles.vendor, profiles.material, profiles.density, profiles.diameter \nFROM selections JOIN spools ON selections.spool_id = spools.id JOIN profiles ON spools.profile_id = profiles.id \nWHERE selections.tool = ? AND selections.client_id = ?'] [parameters: (0, '9e2b4870-df96-11e7-8a76-b827eb362dfe')]
2019-02-03 10:46:56,388 - octoprint.plugins.filamentmanager - ERROR - Failed to fetch selected spools, pause feature will not be available: (sqlite3.OperationalError) no such column: selections.client_id [SQL: u'SELECT selections.tool, selections.client_id, selections.spool_id, spools.id, spools.profile_id, spools.name, spools.cost, spools.weight, spools.used, spools.temp_offset, profiles.id, profiles.vendor, profiles.material, profiles.density, profiles.diameter \nFROM selections JOIN spools ON selections.spool_id = spools.id JOIN profiles ON spools.profile_id = profiles.id \nWHERE selections.client_id = ? ORDER BY selections.tool'] [parameters: ('9e2b4870-df96-11e7-8a76-b827eb362dfe',)]

malnvenshorn, 2019-02-04:
Is this a fresh installation of this plugin or have you upgraded from a previous version?

> no such column: selections.client_id

This sounds like the database wasn't properly created/updated.

kietcallies, 2019-02-05:
I deleted the plugin and reinstalled it.

Here is the error I have getting in the octoprint.log:

2019-02-05 20:05:41,649 - octoprint.plugins.filamentmanager - ERROR - Failed to fetch profiles lastmodified timestamp: Couldn't parse datetime string '1513122926' - value is not a string.
2019-02-05 20:05:41,891 - octoprint.plugins.filamentmanager - ERROR - Failed to fetch spools lastmodified timestamp: Couldn't parse datetime string '1548526211' - value is not a string.
2019-02-05 20:05:42,124 - octoprint.plugins.filamentmanager - ERROR - Failed to fetch selected spools: (sqlite3.OperationalError) no such column: selections.client_id [SQL: u'SELECT selections.tool, selections.client_id, selections.spool_id, spools.id, spools.profile_id, spools.name, spools.cost, spools.weight, spools.used, spools.temp_offset, profiles.id, profiles.vendor, profiles.material, profiles.density, profiles.diameter \nFROM selections JOIN spools ON selections.spool_id = spools.id JOIN profiles ON spools.profile_id = profiles.id \nWHERE selections.client_id = ? ORDER BY selections.tool'] [parameters: ('9e2b4870-df96-11e7-8a76-b827eb362dfe',)]
2019-02-05 20:05:42,139 - tornado.access - ERROR - 500 GET /plugin/filamentmanager/selections (70.91.187.57) 44.61ms

Kiet Callies
Homewaters Web Design

kietcallies, 2019-02-05:
Also:

2019-02-05 19:15:19,388 - octoprint.plugins.filamentmanager - ERROR - Failed to fetch selected spools, pause feature will not be available: (sqlite3.OperationalError) no such column: selections.client_id [SQL: u'SELECT selections.tool, selections.client_id, selections.spool_id, spools.id, spools.profile_id, spools.name, spools.cost, spools.weight, spools.used, spools.temp_offset, profiles.id, profiles.vendor, profiles.material, profiles.density, profiles.diameter \nFROM selections JOIN spools ON selections.spool_id = spools.id JOIN profiles ON spools.profile_id = profiles.id \nWHERE selections.client_id = ? ORDER BY selections.tool'] [parameters: ('9e2b4870-df96-11e7-8a76-b827eb362dfe',)]
2019-02-05 19:15:19,407 - octoprint.plugins.filamentmanager - ERROR - Failed to set temperature offsets: (sqlite3.OperationalError) no such column: selections.client_id [SQL: u'SELECT selections.tool, selections.client_id, selections.spool_id, spools.id, spools.profile_id, spools.name, spools.cost, spools.weight, spools.used, spools.temp_offset, profiles.id, profiles.vendor, profiles.material, profiles.density, profiles.diameter \nFROM selections JOIN spools ON selections.spool_id = spools.id JOIN profiles ON spools.profile_id = profiles.id \nWHERE selections.client_id = ? ORDER BY selections.tool'] [parameters: ('9e2b4870-df96-11e7-8a76-b827eb362dfe',)]

Kiet Callies
Homewaters Web Design

malnvenshorn, 2019-02-05:
> I deleted the plugin and reinstalled it.

What versions? Also I would need a full log which includes the first start of OctoPrint after you installed the plugin. 

The schema of the database doesn't match with the installed version of this plugin. Before version 0.5 the version of the database schema was stored inside the config.yaml. After that it's stored in the database itself. If you upgraded from an old version and the data from the config was some how lost it could result in a unusable state. But that's all speculation without more information.

A solution would be to uninstall the plugin, remove the database and plugin settings from config.yaml. After that you can install it again.

kietcallies, 2019-02-05:
Filament Manager (0.5.3)

Octoprint.log:

2019-02-05 19:14:42,587 - octoprint.startup - INFO - ******************************************************************************
2019-02-05 19:14:42,593 - octoprint.startup - INFO - Starting OctoPrint 1.3.10
2019-02-05 19:14:42,594 - octoprint.startup - INFO - ******************************************************************************
2019-02-05 19:14:42,721 - octoprint.util.connectivity_checker - INFO - Connectivity changed from offline to online
2019-02-05 19:14:43,879 - octoprint.startup - INFO - Blacklist processing done, adding 1 blacklisted plugin versions: roomtemp (any)
2019-02-05 19:14:43,987 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2019-02-05 19:14:51,964 - octoprint.plugin.core - INFO - Plugin Navbar Temperature Plugin (0.9) is disabled.
2019-02-05 19:14:52,852 - octoprint.plugin.core - INFO - Found 58 plugin(s) providing 57 mixin implementations, 81 hook handlers
2019-02-05 19:14:52,858 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2019-02-05 19:14:52,940 - octoprint.server - INFO - Intermediary server started
2019-02-05 19:14:52,941 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2019-02-05 19:14:53,663 - octoprint.plugin.core - INFO - Found 58 plugin(s) providing 57 mixin implementations, 81 hook handlers
2019-02-05 19:14:53,761 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2019-02-05 19:14:55,832 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2019-02-05 19:14:55,864 - octoprint.printer.standard - INFO - Using print time estimator provided by PrintTimeGenius
2019-02-05 19:14:55,874 - octoprint.plugins.nautilus - INFO - 0 device(s) will receive notifications...
2019-02-05 19:14:55,875 - octoprint.plugins.nautilus - INFO - Nautilus for OctoPrint, started.
2019-02-05 19:14:55,939 - octoprint.plugins.cancelobject - INFO - No beforegcode defined
2019-02-05 19:14:55,941 - octoprint.plugins.cancelobject - INFO - No aftergcode defined
2019-02-05 19:14:57,630 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2019-02-05 19:14:57,659 - octoprint.plugins.tracking - INFO - Initialized anonymous tracking
2019-02-05 19:14:57,850 - octoprint.plugins.filamentmanager - ERROR - Failed to initialize database: (sqlite3.OperationalError) duplicate column name: temp_offset [SQL: u'ALTER TABLE spools ADD COLUMN temp_offset INTEGER NOT NULL DEFAULT 0']
2019-02-05 19:15:03,940 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python -m pip" as command to invoke pip
2019-02-05 19:15:09,969 - octoprint.util.pip - INFO - Version of pip is 18.0
2019-02-05 19:15:09,972 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python2.7/site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes
2019-02-05 19:15:09,974 - octoprint.util.pip - INFO - ==> pip ok -> yes
2019-02-05 19:15:09,989 - octoprint.plugin.core - INFO - Initialized 57 plugin implementation(s)
2019-02-05 19:15:10,085 - octoprint.plugin.core - INFO - 58 plugin(s) registered with the system:
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/action_command_prompt
|  Action Commands (0.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_actioncommands
|  Active Filters (0.0.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_active_filters
|  Active Filters Extended (0.0.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_active_filters_extended
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/appkeys
|  AstroPrint (1.2.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_astroprint
|  Auto Collapse Files (0.1.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_autocollapse
|  Autoscroll (0.0.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_autoscroll
|  Backup & Restore (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/backup
|  Bed Leveling Wizard (0.2.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_bedlevelingwizard
|  Bed Visualizer (0.1.7) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_bedlevelvisualizer
|  Cancel Objects (0.2.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_cancelobject
|  Core Wizard (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/corewizard
|  CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/cura
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/discovery
|  Display ETA (1.0.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_eta
|  DisplayZ (0.1.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_displayz
[... cut ...]
|  OctoPrint-IFTTT (1.2.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_IFTTT
[... 364 lines cut ...]
    confirm: Are you sure you want to turn off the printer?
    name: PrinterOff
temperature:
  profiles:
  - bed: 100
    extruder: 240
    name: ABS
  - bed: 60
    extruder: 200
    name: PLA
webcam:
  ffmpeg: /usr/bin/avconv
  snapshot: http://127.0.0.1:8080/?action=snapshot
  stream: /webcam/?action=stream
  timelapse:
    fps: 25
    options:
      interval: 10
    postRoll: 1
    type: timed

I will try that.

Kiet Callies
Homewaters Web Design

kietcallies, 2019-02-05:
After removal:
2019-02-05 23:25:32,086 - octoprint.startup - INFO - ******************************************************************************
2019-02-05 23:25:32,091 - octoprint.startup - INFO - Starting OctoPrint 1.3.10
2019-02-05 23:25:32,093 - octoprint.startup - INFO - ******************************************************************************
2019-02-05 23:25:32,355 - octoprint.util.connectivity_checker - INFO - Connectivity changed from offline to online
2019-02-05 23:25:32,365 - octoprint.startup - INFO - Blacklist processing done, adding 1 blacklisted plugin versions: roomtemp (any)
2019-02-05 23:25:32,471 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2019-02-05 23:25:39,442 - octoprint.plugin.core - INFO - Plugin Navbar Temperature Plugin (0.9) is disabled.
2019-02-05 23:25:40,121 - octoprint.plugin.core - INFO - Found 57 plugin(s) providing 56 mixin implementations, 79 hook handlers
2019-02-05 23:25:40,129 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2019-02-05 23:25:40,211 - octoprint.server - INFO - Intermediary server started
2019-02-05 23:25:40,215 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2019-02-05 23:25:40,944 - octoprint.plugin.core - INFO - Found 57 plugin(s) providing 56 mixin implementations, 79 hook handlers
2019-02-05 23:25:41,046 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2019-02-05 23:25:43,308 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2019-02-05 23:25:43,343 - octoprint.printer.standard - INFO - Using print time estimator provided by PrintTimeGenius
2019-02-05 23:25:43,353 - octoprint.plugins.nautilus - INFO - 0 device(s) will receive notifications...
2019-02-05 23:25:43,356 - octoprint.plugins.nautilus - INFO - Nautilus for OctoPrint, started.
2019-02-05 23:25:43,422 - octoprint.plugins.cancelobject - INFO - No beforegcode defined
2019-02-05 23:25:43,426 - octoprint.plugins.cancelobject - INFO - No aftergcode defined
2019-02-05 23:25:45,206 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2019-02-05 23:25:45,233 - octoprint.plugins.tracking - INFO - Initialized anonymous tracking
2019-02-05 23:25:51,200 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python -m pip" as command to invoke pip
2019-02-05 23:25:57,098 - octoprint.util.pip - INFO - Version of pip is 18.0
2019-02-05 23:25:57,100 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python2.7/site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes
2019-02-05 23:25:57,103 - octoprint.util.pip - INFO - ==> pip ok -> yes
2019-02-05 23:25:57,120 - octoprint.plugin.core - INFO - Initialized 56 plugin implementation(s)
2019-02-05 23:25:57,211 - octoprint.plugin.core - INFO - 57 plugin(s) registered with the system:
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/action_command_prompt
|  Action Commands (0.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_actioncommands
|  Active Filters (0.0.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_active_filters
|  Active Filters Extended (0.0.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_active_filters_extended
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/appkeys
|  AstroPrint (1.2.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_astroprint
|  Auto Collapse Files (0.1.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_autocollapse
|  Autoscroll (0.0.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_autoscroll
|  Backup & Restore (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/backup
|  Bed Leveling Wizard (0.2.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_bedlevelingwizard
|  Bed Visualizer (0.1.7) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_bedlevelvisualizer
|  Cancel Objects (0.2.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_cancelobject
|  Core Wizard (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/corewizard
|  CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/cura
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/discovery
|  Display ETA (1.0.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_eta
|  DisplayZ (0.1.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_displayz
|  Dragon Order (0.1.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_dragon_order
|  EEPROM Repetier Editor Plugin (0.1.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_eeprom_repetier
|  Fan Speed Control (0.2.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_fanspeedslider
[... cut ...] pools.id JOIN profiles ON spools.profile_id = profiles.id \nWHERE selections.client_id = ? ORDER BY selections.tool'] [parameters: ('6c7eb3ea-299e-11e9-af95-b827eb362dfe',)]
2019-02-05 23:37:06,207 - tornado.access - ERROR - 500 GET /plugin/filamentmanager/selections (71.238.5.236) 44.64ms
2019-02-05 23:37:48,564 - octoprint.server.util.sockjs - INFO - Client connection closed: 

I cleared all mention of filament manager in the config.yaml before reinstalling.

Kiet Callies
Homewaters Web Design

malnvenshorn, 2019-02-07:
> I cleared all mention of filament manager in the config.yaml before reinstalling.

As I wrote before you have to remove the database as well.

And please don't post those files as text. It's quite confusing and not really readable. Upload the files instead. Consider using the code formatting for small snippets too.

kietcallies, 2019-02-07:
I apologize; I will follow your reporting guidelines in the future.

May I ask how to remove the database?

Kiet Callies
Homewaters Web Design

malnvenshorn, 2019-02-07:
You can find the database in the data folder of OctoPrint `.../data/filamentmanager/filament.db`

closed 2019-02-12 by malnvenshorn