malnvenshorn/OctoPrint-FilamentManager #143: [Request] Reassign a past print to different spool

jp-powers, 2019-02-17:
I use both Octoprint via the web browser and the Printoid app on my phone. I changed filaments and started a new print via the app and long after it started I checked it via the web app only to be warned the spool would run out... and it definitely did not. Now one spool is showing a negative value and the other hasn't changed.

I know it's more of a "whoops, learning experience I guess" situation. Luckily I put the weight of a printer Slic3rPE estimates in the file names so I can do some basic math and adjust the weight left in Filament Manager accordingly.

However, it made me think of all the other ways external forces could cause the plugin's database to drift away from reality. If there was even a single print history where I could go in and change what filament was printed after the fact it would probably be enough to help in these situations. There's already predictive error messages, but if no error happens the plugin could prompt the user for a "What just happened?" correction.

malnvenshorn, 2019-02-19:
If I understand you correctly you had a wrong spool selected and want to correct the values?

If that's the case you can take a look at the log file. Each print is logged with the calculated amount of used filament. Then you can set the correct values by editing the spools. Of cause you have to do the math on your own.

To prevent such cases I've added a confirmation dialog before each print which can be enabled in the plugin settings. If you bypass this by using a third party app there is nothing I can do. I'm not planning  to extend this plugin with some kind of print history.

Maybe it's time for a FAQ entry. It's not the first ticket of this type.

jp-powers, 2019-02-20:
That's basically the case, and I don't disagree that it's certainly not a necessary issue. All I see in the octoprint.log files are mm usage report:

octoprint.log:|  Filament Manager (0.5.3) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentmanager
octoprint.log.2019-02-14:|  Filament Manager (0.5.3) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentmanager
octoprint.log.2019-02-14:2019-02-14 10:06:12,586 - octoprint.plugins.filamentmanager - INFO - Filament used: 65531.79011 mm (tool0)
octoprint.log.2019-02-15:|  Filament Manager (0.5.3) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentmanager
octoprint.log.2019-02-16:|  Filament Manager (0.5.3) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentmanager
octoprint.log.2019-02-16:2019-02-16 17:24:15,294 - octoprint.plugins.filamentmanager - INFO - Filament used: 31066.51583 mm (tool0)
octoprint.log.2019-02-17:|  Filament Manager (0.5.3) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentmanager
octoprint.log.2019-02-17:2019-02-17 05:40:21,621 - octoprint.plugins.filamentmanager - INFO - Filament used: 47602.87825 mm (tool0)
octoprint.log.2019-02-17:2019-02-17 19:18:44,669 - octoprint.plugins.filamentmanager - INFO - Filament used: 24670.50596 mm (tool0)
octoprint.log.2019-02-18:|  Filament Manager (0.5.3) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentmanager
octoprint.log.2019-02-18:2019-02-18 20:48:02,221 - octoprint.plugins.filamentmanager - INFO - Filament used: 6637.19174 mm (tool0)
octoprint.log.2019-02-19:|  Filament Manager (0.5.3) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentmanager
octoprint.log.2019-02-19:2019-02-19 14:19:36,413 - octoprint.plugins.filamentmanager - INFO - Filament used: 29129.87053 mm (tool0)

Should I be seeing more than that? I don't know the math to figure out the gram usage from that. I don't mind learning, I just don't know where to begin with that. Like I said, luckily I had the Slic3rPE estimated usage in the filename, so I didn't have too much trouble.

I only bring that up since you mention getting a new FAQ entry: Either the FAQ entry should probably detail the conversion math (even vaguely) for those like me, or the log entries could maybe include a bit more data (or, again, I might not be looking in the right place)? If the plugin is letting me work with grams but the logs are working with distance (which makes sense since it far more easily detect that via what Octoprint is doing), then it's already doing that part of the math, right?

malnvenshorn, 2019-02-21:
> Should I be seeing more than that? I don't know the math to figure out the gram usage from that.

No. I decided to log the used filament in mm because it's spool/material independent. The weight calculation is pretty easy:

[image]

> If the plugin is letting me work with grams but the logs are working with distance (which makes sense since it far more easily detect that via what Octoprint is doing), then it's already doing that part of the math, right?

Yes u're right. The plugin sums up the extruded filament length and after the print finishes the value gets computed in grams with the formula above.

closed 2019-02-28 by malnvenshorn