malnvenshorn/OctoPrint-FilamentManager #149: Druck bricht ab wegen Hotend Temperaturfehler

mister2212, 2019-03-02:
**Describe the bug**
wenn das plugin installiert ist bekomme ich beim aufheizen einen fehler wegen der hotend temperatur die dann plötzlich aus irgendwelchen gründen auf über 400 Grad aufheizen will. sobald ich das plugin deinstalliere geht alles ganz normal.

**Steps to reproduce**

**Expected behavior**
Ich starte einen Druck und das Druckbett und und Hotend heizen auf und kurz bevor der Druck startet kommt dann der Fehler weil das Hotend plötzlich auf 400Grad aufheizen will. Egal ob ich manuell aufheize oder einfach den Druck starte. Wenn ich aber nur aufheize und die Temperaturen so laufen lasse ohne dass ich einen Druck starte kommt der Fehler nicht.

**Did the same happen when all other 3rd party plugins are disabled?**
Sobald ich das Plugin Filament Manager deinstalliere funktioniert alles ganz normal und der Fehler ist weg.

**Log file**
[octoprint.log](https://github.com/malnvenshorn/OctoPrint-FilamentManager/files/2922042/octoprint.log)

**Screenshots**

**Additional context**

malnvenshorn, 2019-03-02:
Mit den wichtigsten Punkt hast du ausgelassen **Steps to reproduce**. Mir ist dieses Verhalten noch nicht untergekommen und wenn ich einen Fehler nicht reproduzieren kann kann ich ihn auch nicht beheben. Desweiteren teste ebenfalls noch einmal mit allen anderen Plugins deaktiviert die von dritten stammen, damit ich weiß ob das Problem im Zusammenhang mit einem anderen Plugin steht. 

Zum Schluss noch weitere Fragen: Hast du das Plugin nur installiert oder auch Filamente eingetragen? Falls ja, hast du ein Temperaturoffset eingetragen? Und zu guter letzt: Tritt das Problem auch mit der aktuellen Version von OctoPrint auf?

mister2212, 2019-03-02:
ok sorry hier schonmal die fragen die ich jetzt sofort beantworten kann. das mit den anderen plugins muß ich dann gleich nach dem druck mal testen 

also filamente sind eingetragen und auch offsets. habe auch alle offsets kontrolliert ob ich da vllt einen fehler reingehauen habe aber temperaturen sind alle in dem normalen bereich von 200-245 Grad keine Tippfehler.

OctoPrint 1.3.9 auf OctoPi 0.15.1

malnvenshorn, 2019-03-02:
> also filamente sind eingetragen und auch offsets. habe auch alle offsets kontrolliert ob ich da vllt einen fehler reingehauen habe aber Temperaturen sind alle in dem normalen bereich von 200-245 Grad keine Tippfehler.

Hier könnte schon der Fehler liegen. Es sind Offets, d.h. diese Werte sind relativ zu der eingestellten Temperatur im Gcode zu sehen. Wenn du als Offset also 200° eingegeben hast und die Temperatur im Gcode-File ist ebenfalls 200° ergibt sich eine Temperatur von 400°!

mister2212, 2019-03-02:
ok dann werde ich die ganzen offsets mal löschen und dann nochmal probieren

mister2212, 2019-03-02:
scheint daran gelegen zu haben mit meinem temperaturoffset. jetzt ist der fehler behoben. danke für deine hilfe

closed 2019-03-02 by mister2212