malnvenshorn/OctoPrint-FilamentManager #154: Update to the OctoPi setup - FilamentManager

Stady234, 2019-03-23:
So after much frustration and forum searching.. I finally got the psycopg2 to install.
I am requesting to update the instruction set.
--
Pi client setups
sudo apt-get install libpq-dev python-dev  (<- this will fix the "binary" errors from the development wheel)

sudo ~/oprint/bin/pip2 install psycopg2

malnvenshorn, 2019-03-23:
Yes I will some information to the wiki. But I don't understand the frustration. I quick look at the psycog2 docs would have solved your issue. http://initd.org/psycopg/docs/faq.html#problems-compiling-and-deploying-psycopg2

Stady234, 2019-03-23:
The frustration would be that this in offered on a platform (octopi) that uses a raspberry pi environment.. you only included one line of code to install... which should have worked.. but due to the authors updating code, the single line no longer worked.  While i understand that is not in your control... from what i see in other posts, it gets frustrating that you include dependent programs, but do not make using your plugin "easy" like 98% of all the other plugins.

I can understand that you cant fix every issue, but if this is in "step 2" area... it should be documented and work as intended.  Most other people will not have the time or patience to work through the issues.

And that brings me to the "hours" or research and frustration.  There is little to no documentation that is easily accessible that applies to the general use platform, i.e. raspberry pi.  Most doc is based on Debian or full Linux, and as octopi is stripped to make a "custom" platform... it would make sense to build a plugin that utilizes programs that are either built in or easily updated/installed.

Just my opinion.  Once the program gets installed and working... it is absolutely awesome.

malnvenshorn, 2019-03-24:
I'm trying to make the plugin as easy to use as I  can. But if you have multiple printers and want to share your filament between all instances you have to setup the database and needed dependencies yourself. For this you have to cope with the Linux system and this, obviously, requires some learning. 

From my point of view the big drawback of using pre-made images like OctoPi is that people think everything works out of the box and they don't have to learn how everything works, but this is wrong. If they don't have the time or patience to learn they should look for alternatives, possibly paid ones with official support.

Besides that I'm not running OctoPi or Raspbian on my Raspberry. Therefore I don't know which additional steps might be necessary. If there are some complains because something is not working, you can always write a better/more detailed guide and send it to me.

You said that you've found guides based on Debian. As far as I know OctoPi is based on Raspbian which is basically nothing else than Debian. So finding resources shouldn't be a problem.

malnvenshorn, 2019-03-24:
Edited the wiki page.

closed 2019-03-24 by malnvenshorn