malnvenshorn/OctoPrint-FilamentManager #157: Add support for multiple tools (Prusa MMU/MMU2)

trv-in-3d, 2019-03-24:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

trv-in-3d, 2019-03-24:
Just starting using the PRUSA Multi Material Unit.  It allows printing from up to 5 filaments through a single extruder.      It would be great if Filament Manager would allow be link multiple spools in the database to the gcode tool (T0,T1,T2,T3, etc).    This way a multi-color print using tool 2 and tool 4 would correctly decrement the associated two spools.

This would probably need to be a new "mode" for Filament manager.   Single Filament Mode (as currently written) that requires a single spool to be selected, or Multi Filament Mode, where you have to specific the number of slots, link a spool to each slot, and probably define the GCODE that would be used to activate that spool.  In this example, the GCODE would be the Tx commands - but making this a parameter may enable Filament Manager to work with a variety of other multiple-filament solutions.

malnvenshorn, 2019-03-24:
This plugin supports multiple extruders you just have to configure OctoPrint accordingly. Or do I miss something?

closed 2019-03-24 by malnvenshorn

trv-in-3d, 2019-03-26:
The multi material unit from prusa works as an auto loader/unloader.    Octoprint is correctly configured for the multiple filaments, via a single extruder.   In this setting Filament Manager only offers (correctly) an "extruder 0" option to assign a spool against.

I am asking for a feature to assign spools against gcode "tools", not "extruders".

Gcodes T0, T1, T2 ... will result in different spools being consumed and I cant find a current way that filament manager will track that.

Or am I still missing something?

Thank you for this add-in.  I fully appreciate it's current capabilities and find it very useful.

Sent from my phone. Please pardon the brevity and likely typos.

malnvenshorn, 2019-03-26:
> Octoprint is correctly configured for the multiple filaments, via a single extruder. 

If you can assign only one spool you haven't configured it correctly. If I set the extruder number to two and enable the shared nozzle flag I  assign spools to two tools in the sidebar. This is what you have to do as described in the related ticket.