malnvenshorn/OctoPrint-FilamentManager #159: Connection fail (pgadmin4 and psycopg2 connect script works)

plug-it-in, 2019-04-23:
**Describe the bug**
When I press the Test Connection button for the external database connection it turns red. I can connect to the database as per your setup in postgresql with both pgadmin4 from my Windows machine to the Raspbertry PI OctoPrint, and using a simple script (see screen shot below) connects to postgresql database fine as well.

**Expected behavior**
Button to turn green

**Did the same happen when all other 3rd party plugins are disabled?**
Yes

**Log file**
Drag and drop the octoprint.log here.
[octoprint.log](https://github.com/malnvenshorn/OctoPrint-FilamentManager/files/3108295/octoprint.log)
b
**Screenshots**
[image]
[image]

malnvenshorn, 2019-04-25:
`2019-04-23 15:37:05,022 - octoprint.plugins.filamentmanager - ERROR - Failed to initialize database: No module named psycopg2`

The `psycopg2` module is missing.

closed 2019-04-25 by malnvenshorn

plug-it-in, 2019-04-25:
I am afraid yoiu didnt read the complete log and fixed on the first error. I saw that and fixed that issue already. Otherwise the tiny iof program would not have worked.

plug-it-in, 2019-04-25:
How does one reopen an issue ?