malnvenshorn/OctoPrint-FilamentManager #160: Change Filament WHILE printing

SnorreQWERTY, 2019-04-23:
Hello,

sometimes I change my filament (color) while I'm printing, without interrupting the print process. (like palette 2 without gluing both filaments). Only disadvantage is the missing retract while the cutted filament is in the PTFE.

Therefore it is necessary to change the filament in the software also. But the control is locked while printing. Please open the lock.

DrTechdom, 2019-05-23:
+1 This.... Allow us to include M-code or something into the filament custom g-code of the slicer to allow the color to change automatically which could be caught by the plugin.

pseudex, 2019-08-13:
This option would be great. I have a second extruder. They push Filament through a Y-Joiner into the same nozzle. While Extruder 1 is printing I can change the Filament on Extruder 2 without stopping at all.
On my Colorchange based on gcode I use an extra pause command (M0) after the Filament Change Command (M600).

Another cool Idea would be to have it directly in Gcode to choose the spool. Something like a ID for each spool which can be setup in the slicer.

jp-powers, 2019-09-26:
I'd be interested in something like this as well so I could re-enable the plugin while using my Mosaic Palette 2. I think the Mosaic is arguably a different issue/suggestion from this specifically, but I don't want to pile on different issues unnecessarily. As is I just disable the plugin and ball-park my filament usage, but I've grown to see Filament Manager as a necessity to remove the micro-management of filament usage so it's getting difficult.

I was trying to think thru how it would work with as little modification as possible, but the best that I can figure out is getting the Mosaic slicer (Canvas or Chroma) to insert some g-code on filament changes that Filament Manager could attempt to detect, but I see logistical problems with that. I don't think there's a simple solution.