malnvenshorn/OctoPrint-FilamentManager #162: Multiple printers charge usage to wrong spool

garbled1, 2019-05-10:
**Describe the bug**

I have 2 printers, 2 Pi's both running octoprint with filament manager, hooked up to a single postgres db on an external server.  I noticed on a print the other day, one was printing in yellow, the other in gray, when the gray one finished, it charged itself against the yellow roll, and then the yellow one finished, and charged against the yellow roll as well.  I selected the correct rolls at the start of the print.

I wonder if the issue has to do with my using both instances in the same firefox, in different tabs, and some kind of cookie getting messed up?  not sure here..

**Steps to reproduce**
Print 2 things at once on 2 printers with different filament.  Sometimes it charges the wrong spool.

**Expected behavior**
Not do that.  :)

**Did the same happen when all other 3rd party plugins are disabled?**
Have not tried

**Log file**
do not have log for this event, will see if it happens again and find a log

**Screenshots**
na

**Additional context**
na

malnvenshorn, 2019-05-11:
Just a shot in the dark. Have you copied the config file from one Raspberry to the other? 

The generated client ID in the config must be unique. If you reuse the config without removing that ID you will get the described result.

garbled1, 2019-05-11:
Turns out I did that.. huh.  Let me go see if I can find that config entry and fix it.  Thanks!

closed 2019-05-11 by garbled1