malnvenshorn/OctoPrint-FilamentManager #163: bad temperature set at print start

nicolasloyce, 2019-06-07:
**Describe the bug**
when I'm using filament manager while printing a file from octoprint, temperature is randomly set to 525°c, and it gives to a max temp error. I've found the way to reproduce the bug all the time : 

**Steps to reproduce**
1. upload a file
2. if the job needs for exemple 150g set a spool that is 145g
3. a notification of insuficient spool is displayed
4. print anyway, -> temp is set to 525°c

**Expected behavior**
I'm expected the temperature set in the Gcode : 265°c for me.

**Did the same happen when all other 3rd party plugins are disabled?**
I didnt tryed to disable all my others plugins but when I disable this one + reboot it works again. 

**Log file**
`
2019-06-07 06:10:20,414 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:10:50,468 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:11:20,524 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:11:50,579 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:12:20,635 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:12:50,691 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:13:20,745 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:13:50,802 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:14:20,859 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:14:50,915 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:15:08,527 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-06-07 06:15:20,971 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:15:51,026 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:16:21,079 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:16:51,132 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:17:06,751 - octoprint.server.api - INFO - Actively logging in user user.user from ::ffff:192.168.0.101
2019-06-07 06:17:07,869 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.0.101
2019-06-07 06:17:07,973 - octoprint.server.util.sockjs - INFO - User user.user logged in on the socket from client ::ffff:192.168.0.101
2019-06-07 06:17:08,127 - tornado.access - WARNING - 404 POST /api/plugin/psucontrol (::ffff:192.168.0.101) 12.67ms
2019-06-07 06:17:09,348 - tornado.access - WARNING - 404 GET /plugin/enclosure/updateUI (::ffff:192.168.0.101) 7.83ms
2019-06-07 06:17:17,642 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.0.101
2019-06-07 06:17:21,188 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:17:51,243 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:18:21,299 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:18:51,353 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:19:21,407 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:19:51,460 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:20:21,515 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:20:51,570 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:21:21,625 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:21:51,679 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:22:21,734 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:22:51,790 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:23:21,845 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:23:51,900 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:24:21,954 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:24:52,009 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:25:22,065 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:25:52,121 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:26:22,174 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:26:52,228 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:27:22,282 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:27:52,339 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:28:22,394 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:28:52,449 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:29:22,504 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:29:52,561 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:30:08,531 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-06-07 06:30:22,614 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:30:52,668 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:31:22,723 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:31:52,779 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 06:32:22,834 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
[... 505 lines cut ...]
2019-06-07 07:47:20,836 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 07:47:50,891 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 07:48:20,955 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 07:48:51,008 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 07:49:21,070 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 07:49:51,125 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 07:50:21,180 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 07:50:51,242 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 07:51:07,247 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-06-07 07:51:21,303 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 07:51:51,368 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 07:52:21,422 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 07:52:51,478 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 07:53:21,532 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 07:53:51,594 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 07:54:21,650 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 07:54:51,705 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 07:55:21,769 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 07:55:51,828 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 07:56:21,887 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 07:56:51,942 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 07:57:21,995 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 07:57:52,052 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-06-07 07:58:22,106 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
`

malnvenshorn, 2019-06-07:
Wrong temperature offset?

nicolasloyce, 2019-06-07:
OK, sorry, I didnt want to make it bad, I 've edit my message. 
Yes, you may call it like that so it's a known issue ?

malnvenshorn, 2019-06-07:
I'm referring to the temperature offset you can set in the spool menu (see screenshots). 

There was a similar report of a user who thought that the value is the absolute temperature used for printing instead of an offset. The result was a temperature error (ticket #149).

amemo06, 2019-06-20:
Same for me. Still the step by step is even more simple and no need of an empty filament spool.

- Just select a job to print
- Select a different the filament spool from Filament manager on the main octoprint screen
- Start the print 

Bed will warm up fine
BUT the head heating request will be 400°. And it will really try to reach this temperature thus leading to no way to stop heating. Of course, filament will be burnt. Other issues could occurs if the printer is left unattended which is the main purpose of octoprint

Changing the print job do not help if you succeed due to the above constraint (temperature will not be reached).
Stop the job and restart don't help. Same comment as above
The only solution for me is to shutdown the printer and restart

malnvenshorn, 2019-06-20:
Same question: Do you have a temperature offset set for your selected filament?

amemo06, 2019-06-26:
Since you asked that question, I better understood your previous question and explanations.
Yes I have set the desired temperature per filament instead of a temperature offset.
That, then explains very well the issue I faced. 
Cura sends the requested temperature and Filament Manager adds the offset which was the same temperature thus leading to 2x200° 😭
All seems clear to me now.
Thanks for the explanations. 
I understand quickly if you explain me a long time 😁

malnvenshorn, 2019-07-02:
I guess this can be closed now. Your issue is solved and I haven't received any further answer from the author.

closed 2019-07-02 by malnvenshorn