malnvenshorn/OctoPrint-FilamentManager #165: Link and Description/Note fields on Spools

6cUbi57z, 2019-06-17:
**Is your feature request related to a problem? Please describe.**

It would be great if there were a few more fields available on Spools.  The fields I am thinking of in particular are:

* **Shopping link** (or just "link" to make it more generic) -  This could be a link to the shopping page where more of the spool can be bought from. 
* **Description/Notes** - A text field which can take large amounts of text (wrapped) including new line characters. This could be used for describing the filament, storage instructions/location, special usage instructions, etc.

**Describe the solution you'd like**

In addition to the above, it would probably be a good idea to have a "spool details" dialog which can be opened from a button in the sidebar. This would display the various information fields (exisitng and the above) for the filament in read-only mode for information.

**Describe alternatives you've considered**

Separate information sources (e.g. purchase history).

ruedigernarr, 2019-10-01:
+1  (I was exactly looking for this feature request if it already exists..)

Another sub-feature
* **Date of Purchase** - Showing when the spool was bought to keep track of age of material

stantond, 2019-12-17:
Could I add to this a Notes field for Profiles? It would be a helpful place to store info such as spool weight without filament, so it's easy to weigh the current spool and edit the remaining amount in Filament Manager after forgetting to set it/choosing the wrong spool for a print.

aznoohwee, 2020-04-29:
I would like a notes area too. In my use case I have two printers each with printer specific settings per filament (i.e., linear advance k factor). Rather than storing all these settings in my slicer, it would be nice to have them stored for the filament spool or profile.