malnvenshorn/OctoPrint-FilamentManager #166: Inaccurate Monitoring

kazooless, 2019-06-22:
**Describe the bug**
A clear and concise description of what the bug is.

Spools have a couple hundred grams of filament left when Filament Manager reports it is empty. This has been consistent on the last 8 spools I've used. 

**Steps to reproduce**

Use Filament Manager as you should, choosing the correct spool every print.

**Expected behavior**
A clear and concise description of what you expected to happen.

Expected behavior is a more accurate reporting of the amount of filament used and left on the spool after a print.

**Did the same happen when all other 3rd party plugins are disabled?**
If not which plugin is causing the issue?

This hasn't been tried. 3rd party plugins include Print History, SlicerSettingsTab, Bed Visualizer, Cancel Objects, and Exclude Region.

malnvenshorn, 2019-06-22:
Are the density and diameter settings correct?

kazooless, 2019-06-22:
> Are the density and diameter settings correct?
Yes, I set the density according to the manufacturer's documentation and I measure the filament. It is eSUN PLA+ and I've found the diameter is almost always spot on.

malnvenshorn, 2019-06-24:
Three possible causes come to my mind:
1. Wrong settings
2. Weight is not subtracted from the spool
3. Incorrect calculation of the filament length

To exclude the first two you should check if the filament is subtracted correctly and also compare the weight of the finished print with the value shown using a calibrated scale. If the relative deviation is more or less constant it's very likely an issue with the settings. Otherwise it might be an issue with the calculated filament length, but I don't know if this can be improved. I just took the relevant code from OctoPrints analyzing function.

kazooless, 2019-06-29:
Can you please tell me how it calculates the weight and how it keeps track of the length? Does it keep track as it prints so a failed print is accounted for? Or does it pull the information from the top of the gcode?

Is there any way to see the history so I can tell how much it subtracted from the spool in both weight and length? Or do I just have to make note of the spool status before it prints, and then compare when it is finished?

Thank you for your time,

Jeff

malnvenshorn, 2019-07-04:
> Does it keep track as it prints so a failed print is accounted for?

Yes.

> Is there any way to see the history so I can tell how much it subtracted from the spool in both weight and length? Or do I just have to make note of the spool status before it prints, and then compare when it is finished?

The used filament (in mm) is logged in the `octoprint.log`, but you have to calculate the weight yourself.

kazooless, 2019-08-07:
Helo Sven,

I'm picking this back up now; I got distracted a bit. Sorry about that.

In the logs, this is the only type of entry I see: 
`2019-08-07 13:06:36,521 - octoprint.plugins.filamentmanager - INFO - Filament used: 527.47421 mm (tool0)`

Is there supposed to be anything else logged?
Would it be possible to add a line in the log at the start of each print which records the current status of the spool? (feature request? debug feature?)

Here is an example of the logs when printing. You can see one of the plugins is monitoring every command in case a region is excluded: 

`2019-08-07 09:50:52,750 - octoprint.plugins.excluderegion - INFO - retraction: excluding=False, retract={"e": 6.0, "feedRate": 2700.0, "originalCommand": "G1 F2700 E2691.34382", "type": "RetractionState", "recoverExcluded": false}, returnCommands=[u'G1 F2700 E2691.34382']
2019-08-07 09:50:52,835 - octoprint.plugins.excluderegion - INFO - recovery: excluding=False, originalCmd=G1 F2700 E2697.34382, returnCommands=[u'G1 F2700 E2697.34382']
2019-08-07 09:51:03,578 - octoprint.plugins.excluderegion - INFO - retraction: excluding=False, retract={"e": 6.0, "feedRate": 2700.0, "originalCommand": "G1 F2700 E2704.62817", "type": "RetractionState", "recoverExcluded": false}, returnCommands=[u'G1 F2700 E2704.62817']
2019-08-07 09:51:03,712 - octoprint.plugins.excluderegion - INFO - recovery: excluding=False, originalCmd=G1 F2700 E2710.62817, returnCommands=[u'G1 F2700 E2710.62817']
2019-08-07 09:51:05,383 - octoprint.plugins.excluderegion - INFO - retraction: excluding=False, retract={"e": 6.0, "feedRate": 2700.0, "originalCommand": "G1 F2700 E2705.76432", "type": "RetractionState", "recoverExcluded": false}, returnCommands=[u'G1 F2700 E2705.76432']
2019-08-07 09:51:05,547 - octoprint.plugins.excluderegion - INFO - recovery: excluding=False, originalCmd=G1 F2700 E2711.76432, returnCommands=[u'G1 F2700 E2711.76432']
2019-08-07 09:51:33,035 - octoprint.plugins.excluderegion - INFO - retraction: excluding=False, retract={"e": 6.0, "feedRate": 2700.0, "originalCommand": "G1 F2700 E2732.63715", "type": "RetractionState", "recoverExcluded": false}, returnCommands=[u'G1 F2700 E2732.63715']
2019-08-07 09:51:33,555 - octoprint.plugins.excluderegion - INFO - Event received: event=ZChange payload={'new': 97.2, 'old': 96.96}
`

I'm not a programmer but can sort of get the gist of code when I try to read it. 

Is FilamentManager reading each command in the log, running it through the RegEx filter and then adding the length of filament to the current count? When the print finishes, it logs the total length, makes the calculations and then updates the number displayed in the GUI?

Thank you,

Jeff

malnvenshorn, 2019-08-08:
> Is there supposed to be anything else logged?

Currently no. But you can use it to calculate the weight and compare it to the finished print, see https://github.com/malnvenshorn/OctoPrint-FilamentManager/issues/143#issuecomment-465958071.

> Is FilamentManager reading each command in the log, running it through the RegEx filter and then adding the length of filament to the current count? When the print finishes, it logs the total length, makes the calculations and then updates the number displayed in the GUI?

FM does not parse the log file. It intercepts the GCode send to the printer.

https://github.com/malnvenshorn/OctoPrint-FilamentManager/blob/647af691d6081df2f16d400e834f12f11f6eea56/octoprint_filamentmanager/__init__.py#L378

If a third party plugin modifies the GCode it's probably the cause for the faulty filament measurement you encounter (in this case "Exclude Region"). Disable the plugin and try again.

aharbick, 2021-04-30:
I have been entering the default printed on the rolls (1.75mm) and then using the manufacturer density if I could find it or a googled value (e.g. 1.25g/cc for PLA) and Filament Manager was always running out of filament before my rolls actually were.  Even though I was installing the MMU attachment and having the challenges of tuning it where you waste a lot of filament that filament manager could NOT know about (i.e. my rolls definitely should've run out first)

I decided to test more scientifically.  I started with a fresh roll of Atomic Filament (https://atomicfilament.com/ this is one that I don't know the manufacturer density).  I measured the weight of the filament alone and used 1.75mm for the diameter and printed a fairly large job.

In the octoprint.log I found this entry
```
2021-04-30 11:00:57,335 - octoprint.plugins.filamentmanager - INFO - Filament used: 104883.53678003701 mm (tool0)
```

Then using this formula I computed the grams used:
```
length * pi * radius^2 * density = grams used
```

This is what I got
```104.8835 * 3.1419 * 0.875 * 0.875 * 1.25 = 315.3728``` ​

The amount remaining shown in filament manager was correct by that formula but was off by 25g when I measured the actual weight of the roll.  I figured that the 1.25 density must've been wrong so I rearranged the formula to solve for density given the weight of filament I actually consumed:

```
290 (expected weight used) = 104.8835 * 3.1419 * 0.875 * 0.875 * density
density = 290 / (104.8835 * 3.1419 * 0.875 * 0.875)
density = 1.149
```

That seemed way lower than the default so since I was pretty sure that I hadn't read the weight on my digital scale incorrectly I decided to actually measure my filament.  It measured more in the 1.65-1.68 range.  When I used those numbers the calculation seems much closer to what I was actually measuring.  I've applied the change to my spools in Filament Manager and I expect that it should work.  Had the filament diameter (which is easy to measure) not been off then I would've just adjusted the density as above.

So my recommendations are:

- Measure the diameter of your filament and use actual values
- If that isn't different than the expected value... then carefully measure a before and after weight of your roll, check your octoprint.log for the "Filament used" and compute the actual density of your roll.