malnvenshorn/OctoPrint-FilamentManager #168: Change multiple filament numbering

sticilface, 2019-07-02:
I've just added MMU to my prusa.  The extruders are numbered from 0-4.  Filament manager starts at 1, so II keep selecting the wrong filament!!  Could you add an option to start it at 0?  

Thanks

malnvenshorn, 2019-07-02:
The numbering for the filament selection begins with 0. Can you make a screenshot of what you mean?

sticilface, 2019-07-07:
sorry i meant 1

sticilface, 2019-07-07:
https://www.dropbox.com/s/9dam1vvel5jxccz/Screenshot%202019-07-07%2014.03.41.png?dl=0

sticilface, 2019-07-07:
[image]

malnvenshorn, 2019-07-08:
I won't change that to avoid confusion. OctoPrint counts from 0 therefore the FM too.

closed 2019-07-08 by malnvenshorn