malnvenshorn/OctoPrint-FilamentManager #171: manual update or 2nd manual for octopi image external database

mister2212, 2019-08-04:
**Is your feature request related to a problem? Please describe.**
i think most of the people use the octopi image for octoprint. Is it possible to write a new or 2nd manual to use the external database when you have more printers? Because of the different file path. I can not do it with the normal instructions to get the external database up and running.

**Describe the solution you'd like**
Make a new or 2nd manual with the new file paths for the octopi image.

**Describe alternatives you've considered**

I've tried several times with the wiki instructions and over google trying to find help but there is no manual for the paths in the octopi image

malnvenshorn, 2019-08-04:
According to the README OctoPi is build upon Raspbian therefore the guide from the wiki should work. If you have questions regarding specifics of the image (.e.g. custom paths, etc.) you should ask there or in OctoPrint's support forum.

Because I don't use OctoPi I'm not going to make a separate guide covering the peculiarities of this image. But if someone has the motivation to make one I will publish it in the wiki.

mister2212, 2019-08-04:
Maybe i will try to make a complete new Installation of octoprint with the Wiki and hope that i works

mister2212, 2019-08-05:
when i want to add the database i get an issue

"could not connect to database template1 fatal role pi does not exist"

and when i try to add a user i got this issue

"could not connect to database postgres could not connect to server no such file or directory"

mister2212, 2019-08-05:
i dont understand what i have to write here? My raspberry pi has the ip 192.168.188.44.
which ip should i use here?

Then append the following line to /var/lib/postgres/data/pg_hba.conf

host octoprint_filamentmanager octoprint 192.168.178.0/24 md5
Adapt the IP address to your network, e.g. if your server has the IP 192.168.0.25 use 192.168.0.0/24 instead. This allows all clients in your network to access the database.

malnvenshorn, 2019-08-05:
192.168.188.0/24 

For more information see https://en.wikipedia.org/wiki/Subnetwork

mister2212, 2019-08-05:
ok and in the plugin under adress i write the ip of raspberry?

mister2212, 2019-08-10:
ok on the server it works but the client my 2nd pi cannot connect to the database. what i have to do or install on the 2nd pi that he can connect and use the database? can you help me again please?

Stady234, 2019-08-10:
Look at what i did on my issue to see if that helps.. #172

mister2212, 2019-08-11:
Thank you very much Stady234 now everything is working.

closed 2019-08-19 by mister2212