malnvenshorn/OctoPrint-FilamentManager #172: Setup PostgreSQL on Raspbian (Stretch) -- Connection will not pass test

Stady234, 2019-08-09:
**Describe the bug**
Followed the setup guide.  Last step to click "test" in Octoprint, and button stays red.

**Steps to reproduce**
click "test" in Octoprint, and button stays red.

**Expected behavior**
Button to turn green.

**Did the same happen when all other 3rd party plugins are disabled?**
Yes

**Log file**
[octoprint.log](https://github.com/malnvenshorn/OctoPrint-FilamentManager/files/3488186/octoprint.log)

**Screenshots**
[image]

**Additional context**
I have been experiencing many "non-connects" with the ArchDB.  Not sure if Octoprint is cause or recent Win10 updates due to running Arch in Virtual.  Has been working for months.  Wanted to try the raspian way due to the previous issues.

Stady234, 2019-08-10:
Had to make other changes to the prostregsql config files than what you have listed, but finally got it working.

> In the /etc/postgresql/9.6/main/postgresql.conf change the listen_addresses to

- [ ] removed the "#" in the front of the - listen_addresses = '*'

- [ ] Add "#" to all ssl entries (to align with archbd version installed a few months ago)

> Then append the following line to /etc/postgresql/9.6/main/pg_hba.conf

- [ ] Change the "md5" on the - host octoprint_filamentmanager octoprint 192.168.178.0/24 md5 to "trust" like so::
host octoprint_filamentmanager octoprint 192.168.178.0/24 trust

You may or may not need the ssl change, but it is working on all 4 of my octoprint

malnvenshorn, 2019-08-10:
> removed the "#" in the front of the - listen_addresses = '*'

Of course, otherwise the option has no effect.

> Add "#" to all ssl entries (to align with archbd version installed a few months ago)

What is `archbd`?

> Change the "md5" on the - host octoprint_filamentmanager octoprint 192.168.178.0/24 md5 to "trust"

If it works with `trust` and not with `md5` then you should check your password. `trust` should only be used for `localhost` connections.

I'm also closing this, because you solved your issue.

closed 2019-08-10 by malnvenshorn