malnvenshorn/OctoPrint-FilamentManager #177: Barcode scanning/printing

Wirepower, 2019-10-04:
a feature I’d like added. 
To use a barcode scanner (considering most spools already have a barcode that’s unique on them) and if not, printing one on the web is very easy and just attach it to your spool.
This will retrieve details of that spool on your database otherwise if it’s not there it would allow you to create a new entry for it.

USB barcode scanners are cheap. However I understand not everyone wants a barcode scanner so the ability to manually enter barcode numbers could be achieved as well.

The info once scanned could display everything needed to know about the brand/colour/size/temperatures/amount of spools/etc

I’ve made a simaler system on MS Access and it is very useful. Would be even better on rpi/octoprint.

ruedigernarr, 2019-10-05:
+1

@Wirepower do you just use the (2D) barcode to store an ID of the spool which refers to a database entry of the respective spool or so you want to store more info in the (eg 3D QR) barcode directly?

Wirepower, 2019-10-05:
2D barcode as most spools already have a barcode,
All I’d be doing is using the existing barcode on the spool to add to my database with all details I would manually put such as bed temp, nozzle temp for that particular product.

This way if I purchase another exact same spool, I already have the details in my database. 

I don’t want QR as business don’t use them when they sell a product but use a normal barcode which also has numbers below it. 
This is ideal as it allows the user to manually input the barcode number if they don’t want to use a barcode scanner.

Hope that clears it up

Wirepower, 2019-10-05:
What I’m basically trying to achieve is this.
Have a database using the existing barcodes on product spools as this makes the number unique to that particular product and it’s already there on the product (usually)
This way when I scan it in, I can’t add it to the database and input all the details of this product as well as recommended bed/nozzle temperatures etc.
Now if I buy other products I would do the exact same, later on I may purchase the exact same product again so I would scan it and it would bring up the details of that product with the require temperatures instead of me searching for them. 
The more brands/material types I use will build the database up initially as I would of manually inputting the details on my initial purchase.

Because people all over the country don’t have access to specific brands, it would be ideal as the database would be filled with products the user would use or have access to.

Now because there are some spools which don’t have a barcode, embedding a manual barcode the spool could be done by either web based barcode generator which would allow you to print to a sticker to stick to the spool. 

I have basically made this program using MS access however it isn’t ideal as the user would require ms access to utilise it.
So I’m in the process of making a c# program that would have everything embedded in the software.

In future I could generate a massive database from users throughout the world and have one big database but this would be a bit hard when the user generates a barcode as it doesn’t become a unique number for that product but could be possible as then I could list the database is sql server that everyone could have access to. 

Now the point I’m getting at is, if everyone uses octoprint, why not embed the software onto it instead of making a seperate c# program which would require a pc, when a raspberry pi is more then capable and is with the printer anyways :)

ruedigernarr, 2019-10-05:
Get it, makes totally sense. Also very simple - just use the (in many cases already existing ) 2d barcode ID as the ID for the filament manager spool plugin database entry and make it easier to connect without using the drop-down and search for the right spool. 

Here are also other feature requests for more input fields already reported, which would add up to the functionality you describe.

ruedigernarr, 2019-10-05:
I think what is technically needed is:

* Another database field for an ID (which can be used defined) 
* A search field on to of the spool drop-down to use that for scanning, pressing enter would select the spool.

Since you can also create labels yourself you can attacg special characters like "pressing enter" to the ID in the barcode and just scanning the code would select and login the right spool.

Wirepower, 2019-10-05:
What you think of the idea?

oschwartz10612, 2020-07-04:
Dredging up an old thread here. I recently had this idea and implemented some basic functionality using the API here: https://github.com/oschwartz10612/Filament-Manager-Barcodes. Excuse the code; python is not my comfort zone and it a proof of concept.

It works but the main issue is that the UI does not update on API calls. Any idea what we could do to remedy this? New to octoprint plugins.

oschwartz10612, 2020-07-08:
Possible fix in #218