malnvenshorn/OctoPrint-FilamentManager #178: Support for Palette2

audiorazor, 2019-10-14:
Not sure if its possible but I would love to see Palette filament tracking implemented.  I believe because it doesnt use tool changes that it will not work by default.

malnvenshorn, 2019-10-15:
> doesnt use tool changes

How does it work then? For me it looks like a normal shared nozzle setup.

audiorazor, 2019-10-16:
It is the normal shared nozzle setup but for some reason octoprint only sees the print as a single color.   I can send you a processed file and see if you can spot the differences.

jp-powers, 2020-01-16:
Jumping in because I'm very interested in to see Palette's supported as well.

Mosaic Palette 2's work in two says: with or without a "Canvas Hub." If you're trying to us this plug in with it, you're almost certainly using a Canvas Hub, which is just a Raspberry Pi with Octoprint and some special plugins. They supply the plugins if you want to supply your own Raspberry Pi. There's no point in this plugin supporting the without a hub option, in my opinion.

The slicer, Canvas or Chroma (Canvas is their own slicer that's browser based, Chroma is their older slicer that processes gcode from another slicer to work with the Palette), will generate non-gcode data to tell the Palette when to do it's thing (which boils down to cutting a filament, heating it up, then smooshing it into another filament). This additional data is basically just inserted at the beginning of the gcode file and throughout the gcode, and their plugin interprets the additional gcode and tells the Palette 2 when to do splices.

When I tried using Filament Manager with imy Palette I think it can't pick it up because the Palette isn't supplying any "traditional" g-code to the "tool" changes. I've attached an .mcf.gcode file from Canvas as an example (Changed to .txt so github will accept it, but the mcf part of the file extension designates the Mosaic differences versus traditional gcode which I'm fairly sure is how the Mosaic plug-ins tell the difference between something it should handle and something it should just let Octoprint do on it's own), and to hopefully make it easier to figure out what's it doing here a section of the filament change I believe:

```
; *** Selecting and Warming Extruder 4 to 215 C ***
M104 S215 T0
; Chroma toolchange handling
; toolchange at totalE = 1026.49463 mm
; changing logical extruder from T2 to T3
; Reset extruder pos
G92 E0
; chroma inserting next dense tower segment
; move to transition tower
```

And another:

```
; *** Selecting and Warming Extruder 3 to 215 C ***
M104 S215 T0
; Chroma toolchange handling
; toolchange at totalE = 1363.99695 mm
; changing logical extruder from T3 to T2
; Reset extruder pos
G92 E0
; chroma inserting next dense tower segment
; move to transition tower
```

I say "I believe" there because I'm not well versed enough in these things to really be certain. Admittedly, that "I believe" statement should apply to all of this as it's just what I can figure out from my relatively lay-man's understanding. Hopefully this helps! I imagine it wouldn't be super easily implemented, but I'd love to see Palette's supported as these burn so much filament on transition towers it's hard to really tell what is consumed, and I miss using this plugin.

[Baby-Yoda-MP.mcf.gcode.txt](https://github.com/malnvenshorn/OctoPrint-FilamentManager/files/4069362/Baby-Yoda-MP.mcf.gcode.txt)

malnvenshorn, 2020-02-04:
> When I tried using Filament Manager with imy Palette I think it can't pick it up because the Palette isn't supplying any "traditional" g-code to the "tool" changes.

Yes you are right. There is a comment `; changing logical extruder from T3 to T2` but the actual command for selecting the tool `T3` is missing. I don't know how this Palette thing works and how it knows which extruder to use, because there is no command in the gcode. The FM plugin analyses the gcode and the only tool in use it sees is `T0`.