malnvenshorn/OctoPrint-FilamentManager #179: Support for MariaDB as back end Database

Snille, 2019-11-02:
**Is your feature request related to a problem? Please describe.**
No problem really, just a nice to have feature, it would be nice to be able to use MariaDB as a back end database. I have other systems using my "House-database" that I have on a "database-server". Would be nice to be able to utilize that database for Filament as well. 

**Describe the solution you'd like**
Just a way to select MariaDB as a back end database.

**Describe alternatives you've considered**
I was thinking of installing an instance of PostgreSQL on my database server, but rather not. 

**Additional context**
For the moment I have two Octoprint instances that utilizes the PostgreSQL on the local Raspberry.  But I'm afraid that I'll lose the data if the Pi SD-Card crashes for example. I would rather have the "data" on a safe server. :)

Thank you!

malnvenshorn, 2019-11-04:
The reason that there is no support for MySQL/MariaDB is that these database systems doesn't have a notification mechanism. There is a request for this on the [MariaDB tracker](https://jira.mariadb.org/browse/MDEV-16590).

Snille, 2019-11-04:
Ah, I see. Thank you for letting me know. I'll be installing a postgresql along side my Maria then. :) It's not a huge problem, just have to remember to backup them both. :)

closed 2019-11-04 by Snille