malnvenshorn/OctoPrint-FilamentManager #188: Can't connect to Postgres database

ithelpme, 2020-01-08:
**Describe the bug**
I can't use postgres database.  Specifically, in the octoprint log, I'm seeing No module named psycopg2.  It is in fact installed using the path applicable to my install with is Octoprint Stretch so the path ~/oprint/bin/pip .  I've followed many guides but no luck.  I've started all over using a brand new SD card to install Octoprint stretch and followed line by line on several guides and I can't get the Test Connection button to turn green.  I installed Postgres on the raspberry pi itself and also tried using the postgres database on my UnRaid server.

**Steps to reproduce**
Install Octoprint Stretch from scratch to Raspberry Pi 3B +.  Followed this guide precisely: https://community.octoprint.org/t/filament-manager-plugin-how-to-setup-postgresql-database/10698

**Expected behavior**
Button to turn green

**Did the same happen when all other 3rd party plugins are disabled?**
No other plugins are installed other than Filament Manager

**Log file**
Drag and drop the octoprint.log here..  Attached
[octoprint (5).log](https://github.com/malnvenshorn/OctoPrint-FilamentManager/files/4036408/octoprint.5.log)

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

ithelpme, 2020-01-08:
Responding to my own post.  Octoprint buster light or version .17 works fine so it must be some sort of interoperability with .16 and the psycopg2 module.  I guess I will need to install .17 on all of my raspberry pi.  Using Octoprint .17, I was able to connect to a Postgres on my UnRaid server without any issue.

malnvenshorn, 2020-02-04:
Great that you figured it out.

closed 2020-02-04 by malnvenshorn

hermy65, 2020-06-22:
@ithelpme Im running into issues getting this to work with a postgres container on my unraid as well. Any input on what you did by chance?