malnvenshorn/OctoPrint-FilamentManager #195: Unable to select spool; have to close and reopen 2nd time to select.

Silventar, 2020-02-13:
**Describe the bug**
When I start a print from the Octoprint file manager, it creates a popup to allow me to select a spool. However, the spools drop down is not shown. About one second later, in the background, Octoprint shows a "frame size calculated" message in the top right of the screen that lasts for a couple seconds. If I close the Filament Manager popup click the print again, the new popup shows the drop-down.

This seems to only happen if I send the gcode from my slicer and move to Octoprint immediately to start the print. If I wait a minute or two after sending the gcode to start the print, the issue does not happen.

**Steps to reproduce**
1. Go to the Octoprint File Manager in the sidebar, press the print icon for the file to print.

**Expected behavior**
The Frame Size Calculated message should display, then the Filament Manager popup should display to select the spool of choice.

**Did the same happen when all other 3rd party plugins are disabled?**
Yes.

**Log file**
[octoprint.log](https://github.com/malnvenshorn/OctoPrint-FilamentManager/files/4199141/octoprint.log)

**Additional context**
Raspberry Pi 4B, 4mb RAM
OctoPrint: 1.3.12
Filament Manager: 0.5.3

NameOfTheDragon, 2020-03-01:
I'm also seeing this. I like being prompted to confirm the spool but usually the first time all the selection list and buttons are disabled so you have to cancel and hit Print again, then it will usually work.

adamsiem, 2020-04-01:
This is still an issue on my new install of OctoPi 0.17.0 with OctoPrint 1.4.0.

I am unable to print anything unless the Filament Manager is disabled. As @NameOfTheDragon described, I cannot print, and even clicking Cancel does not work - as the Print button is greyed out for my model until I restart OctoPrint without the Filament Manager plugin.

adamsiem, 2020-04-03:
@malnvenshorn Could this bug (inconsistent ability to 1. select spool and 2. click active 'print button' - that is not greyed out) be something to do with modal confirmation - controlled by the framework?

Here is the **modal to confirm spools**: https://github.com/malnvenshorn/OctoPrint-FilamentManager/blob/master/octoprint_filamentmanager/templates/spool_confirmation.jinja2

For example...

Is it that the event is not consistently sent to allow the 'print' button to be activated? 

I see the print button is waiting on this to be enabled as part of the data-bind:

`event: { change: function() { $root.viewModels.confirmation.checkSelection();`

posicat, 2020-04-13:
I ran into this on my machine, looks like it was just a case of  the database not being online when the Octoprint server got started.  I put postgresql in as a dependency on my octoprint service and it worked fine after that.

adamsiem, 2020-04-13:
When you saw the error, did you have no spool toggle, or a toggle and you were unable to select ‘print’?

FixxCZ, 2020-04-18:
Having the same issue on OctoPi 0.17.0 with OctoPrint 1.4.0. and Filament Manager 0.1.4. I have to click cancel go to dropdown and select none in the Tool 0. Then it allow me to choose it.

FixxCZ, 2020-04-19:
Resolved it in Brave browser by disabling the shield. No problem since then.