malnvenshorn/OctoPrint-FilamentManager #197: Separate Spool Selection when using common database

Wob76, 2020-03-04:
I am currently using PostgreSQL as common database for 2 OctoPrint instances, this is working fine except if I change the active spool on either instance it changes on the other, even if the other unit is mid print. As a result I can't run both instances at the same time running different spools, I can't see any option that I am missing so I assume this needs to be a future request?

Thanks,
Wob

tech-kev, 2020-03-06:
I had the same issue and I found here the solution: https://github.com/malnvenshorn/OctoPrint-FilamentManager/issues/120

The `config.yaml` is located at `/home/pi/.octoprint/`. Just search in the file for `clientID` and change it.

Wob76, 2020-03-07:
Great, thanks for the heads up, I'll give that a try but sounds like it's my problem. I did do a restore from one pi to the other so very likely the ID followed.

Wob76, 2020-03-11:
Finally got back to this, that did the trick, thanks again @tech-kev closing this off.

closed 2020-03-11 by Wob76