malnvenshorn/OctoPrint-FilamentManager #232: Unable to connect to external database

blindguy, 2021-01-24:
**Describe the bug**
Trying to connect the filament manager to a external database and I keep getting this message from three different octoprint pi's.
the 192.168.1.1 is my local PC IP address.  The postgresql DB is sitting on my server and I can access it via DBever.
WARNING - 400 POST /plugin/filamentmanager/database/test (::ffff:192.168.1.1) 21.05ms
they docs say to run this:
~/OctoPrint/venv/bin/pip install psycopg2 but that file/direcotry does not exist for me:
root@octopi03:/home/pi# ~/OctoPrint/venv/bin/pip install psycopg2
bash: /root/OctoPrint/venv/bin/pip: No such file or directory

**Steps to reproduce**
1. run this:
~/OctoPrint/venv/bin/pip install psycopg2 

**Expected behavior**
I expect the plugin to connect to the DB

**Did the same happen when all other 3rd party plugins are disabled?**
Everything else works fine

rufinus, 2021-01-30:
try Steps 1 - 7 from https://community.octoprint.org/t/filament-manager-plugin-how-to-setup-postgresql-database/10698

I use it with an postgresql docker container on one of my servers

Gradius802, 2023-05-17:
So when I setup filament manager and added a database in the use external database page, I clicked the test button and it succeeded, upon reboot I got the error "filament manager initializing..." in the same page, and none of my spools loaded.

I was able to fix it by doing the following: (if you're using a Raspberry Pi)

sudo raspi-config

1. System Options

S6 Network at Boot