malnvenshorn/OctoPrint-FilamentManager #233: Cant load Filament manger after new update from octoprint

bfk220, 2021-01-30:
Running octoprint 1.5.3 and filament manager disappeared.  Reload from repository does nothing.

rufinus, 2021-01-30:
same here.  installing from console shows no errors or warnings (besides pip version) are thrown.

grojashc, 2021-01-31:
Same here Octoprint 1.5.3

ruralaccent, 2021-03-07:
Me too. It's totally disappeared.

MrEditor97, 2021-07-11:
I had this problem earlier, it seems to be to do with the "sqlalchemy" package.

However, I found a workaround this morning:
Find the file called **"compat.py"** within "**_[python location]_/site-packages/sqlalchemy/util/**" folder

Replace **time.clock()** with **time.perf_counter()**.

Restart Octoprint and then Filament Manager should function as expected again.