malnvenshorn/OctoPrint-FilamentManager #238: Spool Selection Dialog when Spool is already Correct

GoodyGizmos, 2021-09-17:
**Is your feature request related to a problem? Please describe.**
It's a small nuisance to have to reselect the already correct spool before starting a print.

**Describe the solution you'd like**
I just started using the "Show dialog to confirm selected spools before starting/resuming the print job" and I mostly like it, except can you change it to not require selecting the same filament from the dropdown if it's already correct? Maybe add a "Continue" button and/or preselect the same spool in the dropdown?

**Describe alternatives you've considered**
Dealing w/ it ;-)

**Additional context**
N/A

[image]

geeklimit, 2022-01-23:
Can confirm this is happening to me as well. "There are no information available about the active tools for the print job" when a valid spool is already selected.  Clicking 'cancel', then restarting the print brings up the dialog box as expected, and 'Start print' is able to be clicked. Happens on every print.

GitHubGoody, 2022-02-18:
> Can confirm this is happening to me as well. "There are no information available about the active tools for the print job" when a valid spool is already selected.  Clicking 'cancel', then restarting the print brings up the dialog box with the current spool selected, as expected, and 'Start print' is able to be clicked. Happens on every print.

I agree this is an issue, but I think it's separate from my request.